LibrePCB/LibrePCB #396: Use a default component style on schematic & component selector

hooperbloob, 2018-12-19:
I love LibrePCB's capability to present alternate renderings for components, i.e. 
American vs European component symbols.  Can users select their preferred 
style as a preference option to show on the component selector and the schematic
itself?  Maybe have the option to choose the preferred style when the library is
first loaded.

dbrgn, 2018-12-19:
Hi, thanks for the suggestion!

I think this is something that's already planned. @ubruhin did you intend to use the "norm" field for this? In that case we would probably need to standardize the values somehow.

Are there other symbol variant choices besides IEEE (US) vs IEC (EU)?

Maybe the design of such a feature should be discussed at https://librepcb.discourse.group/ since there are still quite a few open questions.

hooperbloob, 2018-12-19:
Glad to hear.   Not being able to render schematics with your preferred symbols is like dealing with the loss of favourite words from your vocabulary.

ubruhin, 2018-12-20:
Yes, this is intended, and parts of it are already implemented. In the workspace settings you can choose your preferred norm(s):

[image]

But as far as I remember, it's not yet used at all :grin:

PR #430 (2019-03-17): AddComponentDialog: Pre-select symbol variant by project norms
- Enable project settings menu item in board/schematic editors (I guess they were hidden by accident).
- `AddComponentDialog`: Pre-select symbol variant by project norms

Now you can set your preferred norm in the project settings dialog:
[image]

And then the "Add Component"-dialog automatically pre-selects the component symbol variant according the specified norm:
[image]

Fixes #396
closed 2019-03-17 by ubruhin