LibrePCB/LibrePCB #399: High DPI Display Issues

NZSmartie, 2018-12-26:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.0
Git Revision:     d7458d3b3e126499902e1a66a0ef889f516a7c97
Build Date:       2018-11-24T16:25:14
Qt Version:       5.9.7 (built against 5.9.7)
CPU Architecture: x86_64
Operating System: Windows 10 (10.0)
Platform Plugin:  windows
```

##### SUMMARY
On Windows 10 with the Display Scaling set to 200%. Creating a new Package in Library Manager, the Package dialog will prompt for the Package Pads and has a visual bug as seen in the following screenshot.
[image]

My laptop has a High DPI display with Windows 10 configured to use a 200% scaling. This does caused a few visual elements to be rendered incorrectly, however, in LibrePCB, this is the first case I've seen so far.

##### STEPS TO REPRODUCE
1. Setting Windows to 200% scaling
   1. In Windows 10, navigate to System Settings.
   1. Within the Settings window. Navigate to System -> Display -> Scale and Layout.
   1. Set the scaling to 200%
   1. Log out of Windows and log back in again.
1. Observing the Visual Bug
   1. Start LibrePCB
   1. Open the Library Manager
   1. Start the New Package dialog.
   1. On the Package Pads step, observe the visual bug

##### EXPECTED RESULTS
To easily read the text and continue with the New Package dialog

##### ACTUAL RESULTS
A visual bug is preventing the user from reading text in the dialog.

ubruhin, 2018-12-27:
In that particular case, we could just remove the first column of the table as we don't want tho show UUIDs anyway (see #337 ) :wink:

dbrgn, 2018-12-27:
Yes, but there seems to be an issue with Qt scaling of the widgets. The heights aren't hardcoded, right?

NZSmartie, 2018-12-27:
>The row height is set to the height of the buttons in the last column, and that button height is hardcoded...

I'm seeing this is more places. Almost every table with a button in it's rows is set this way. 

[image]

Does Qt support setting a DPI independent height?

P.S. On that note, i'll have a look myself, Now that i'm trying to use LibrePCB for a project, would be a good time to learn how to fix bugs I find too.

ubruhin, 2018-12-27:
> I'm seeing this is more places.

True...

> Does Qt support setting a DPI independent height?

I have no idea. Actually I think even a DPI independent height is not a very elegant solution, although it might work ;) Maybe the buttons could be made a bit smaller with a custom stylesheet or so. I mean, generally the issue with high DPI should be solved automatically when using the default button size. Then we only have the issue that tables are not space efficient, but that might be fixable in a different way.

> P.S. On that note, i'll have a look myself, Now that i'm trying to use LibrePCB for a project, would be a good time to learn how to fix bugs I find too.

Cool! :+1:

rnestler, 2019-01-15:
Qt actually documents how to handle high DPI displays: https://doc.qt.io/qt-5/highdpi.html

ubruhin, 2019-03-14:
@NZSmartie Could you maybe check if the following nightly build has the high dpi display issue fixed?

https://download.librepcb.org/nightly_builds/399-fix-highdpi-table-cell-issues/librepcb-nightly-windows-x86.zip

PR #429 (2019-03-17): Fix cell height issues of several tables on high DPI displays
I hope this fixes #399 by stop using hardcoded table cell heights, but I can't test it as I don't have a high DPI display. Testers welcome ;)

Fixes #399 (hopefully)
closed 2019-03-17 by ubruhin