LibrePCB/LibrePCB #406: Improve default layer colors

ubruhin, 2019-01-14:
##### SUMMARY

The current default layer colors are following (small demo project created [here](https://github.com/LibrePCB-Projects/layer-colors)):

[image]

Concrete issues:
- Many layers have exactly the same color.
  - Placement, names and values (which are typically used for silkscreen) cannot be distinguished from documentation layers. Thus it's not visible what's printed on silkscreen and what not.
- Bottom copper has very bad contrast to the background and thus is not very readable.

##### SOLUTION

Find good layer colors and adjust `libs/librepcb/common/graphics/graphicslayer.cpp` accordingly. The small test project mentioned above can be used to test the layer colors. 

Note: It's probably easier to play around with the layer colors directly in the project's file `boards/default/settings.user.lp` instead of adjusting the source code and compile after every change. The layer colors in the source code are only used as default anyway, if the board settings file exists, those layer colors will be used instead.

dbrgn, 2019-01-14:
Maybe also add a link to this to the discussion forum :)

shakenov-chinga, 2019-01-15:
Hi, everybody!
Maybe only two colors for inner copper layers can be used: [image] and [image], or any other colors, one for even and the other for odd numbered layers. The reason is that for 4 copper layers board looks like a mess already, and there is no way to see a clear look without hiding some of them. Also, this way colors that are used in inner layers will be freed, and can be used elsewhere instead. Outline for text layers (names and values) and for airwire ("Guide") can be used too, if it is relatively easy to implement.

ubruhin, 2019-01-27:
> Maybe only two colors for inner copper layers can be used: 7fffff and 7f00ff, or any other colors, one for even and the other for odd numbered layers.

Yeah that would be an option, although it would be nice to have unique colors for each layer, but I see the problem with the limited count of available colors :grin: 

Feel free to use the demo project I mentioned above to play around with the colors and post a screenshot to review :)

dbrgn, 2019-03-17:
Here's a proposal.

Before/after default layers:

[image]

Before/after all layers:

[image]

Main changes:

- New shades for top/bottom copper (less aggressive colors)
- Silkscreen is shown in full white
- General comment/documentation layers are shown in gold
- Board outline is light blue
- Inner layers all have the same base color, but in 5 different alternating shades
- Solder paste and mask layers are much more transparent
- Names and values have the same color on top and bottom, but names are slightly green-ish and values are slightly blue-ish.

Here's the file, if you want to play around with it:

[settings.user.lp.draft1.txt](https://github.com/LibrePCB/LibrePCB/files/2975782/settings.user.lp.draft1.txt)

These are the colors used (and opacity values):

```
red:    #cc0802
blue:   #354ea0
green:  #6db515
lime:   #808000
gold:   #e59500
yellow: #ffff00
ruby:   #b21056
purple: #b2109f
white:  #ffffff
grey:   #e0e0e0
teal:   #00ffff

top_cu: red/96
bot_cu: blue/96
brd_vias_tht: green/96
brd_pads_tht: green/96
in00_cu: purple/96
brd_outlines: blue/ff
brd_milling_pth: blue/ff
brd_comments: gold/b4
brd_documentation: gold/b4
brd_alignment: gold/b4
brd_airwires: yellow/ff
top_placement: white/dd
bot_placement: white/dd
top_documentation: grey/96
bot_documentation: grey/96
brd_sheet_frames: grey/96
top_stop_mask: white/20
bot_stop_mask: white/20
top_solder_paste: grey/20
bot_solder_paste: grey/20
top_courtyard: teal/46
bot_courtyard: teal/46
brd_guide: lime/ff
brd_measures: lime/ff
```

Note: Highlighted colors are not yet implemented, I don't want to do the work twice so I'd welcome some feedback first :slightly_smiling_face: 

Note2: The contrast of the copper layers against the background is still quite low. I think it looks better that way, but for people with visual disabilities or bad screens we could also add a high contrast mode/theme.

ubruhin, 2019-03-19:
Looks much better than the old colors! :+1:

My thoughts:
- Board outline and milling are basically the most important layers (they define the shape of the PCB), thus IMHO they should be much more visible. I would propose white (I think it doesn't hurt if it's the same color as the placement layers).
- The visibility of bottom copper is still not very good (low contrast to background). Maybe use a color other than blue?
- Solder paste looks very transparent, maybe make it a little bit less transparent. Stop mask maybe too.
- Inner copper layers are very similar. It would be nice to have better distinguishable colors at least for the first two inner layers. Then we would have 4 different copper layer colors, which should be enough for most projects.

dbrgn, 2019-03-28:
> Board outline and milling are basically the most important layers (they define the shape of the PCB), thus IMHO they should be much more visible. I would propose white (I think it doesn't hurt if it's the same color as the placement layers).

Done.

> The visibility of bottom copper is still not very good (low contrast to background). Maybe use a color other than blue?

I like blue. An alternative would be green (like KiCAD). But let's try making the blue lighter first.

What about this? I'd say red and blue both have about the same contrast.

[image]

> Solder paste looks very transparent, maybe make it a little bit less transparent. Stop mask maybe too.

The reason is that when they're combined, the opacity adds up, especially for THT pads.

Top/Bot mask on the left, only Top on the right:

[image]

I think those layers should be quite light, and usually you probably won't have them enabled anyways. But if you think they should be stronger, this is 50% more opaque (0x30 instead of 0x20). What do you think?

[image]

> Inner copper layers are very similar. It would be nice to have better distinguishable colors at least for the first two inner layers. Then we would have 4 different copper layer colors, which should be enough for most projects.

What about using ruby (similar to red / top) and purple (similar to blue / bottom) in alternating medium / light / dark shades?

Here would be the first 6 layers, after that the colors would repeat.

[image]

ubruhin, 2019-03-28:
> Done.

Milling is still blue ;)

> What about this? I'd say red and blue both have about the same contrast.

Hard to say with just looking at the screenshot, but the contrast of blue looks still not very high... What about something like cyan?

> But if you think they should be stronger, this is 50% more opaque (0x30 instead of 0x20). What do you think?

I like it :+1:

> What about using ruby (similar to red / top) and purple (similar to blue / bottom) in alternating medium / light / dark shades?
> Here would be the first 6 layers, after that the colors would repeat.

Yep, looks better! :+1:

dbrgn, 2019-03-28:
> Milling is still blue ;)

Ah sorry, missed that one. Updated.

> Hard to say with just looking at the screenshot, but the contrast of blue looks still not very high...

I think it's quite good. At least as good as red! Here's the updated settings file. I think it should be mostly done (including highlight colors).

[settings.user.lp.draft3.txt](https://github.com/LibrePCB/LibrePCB/files/3020420/settings.user.lp.draft3.txt)

By the way, to generate the highlighted colors, I generally entered the color in the generator of https://coolors.co/. Then in the column with that color you click on the "alternative shades" icon and choose the color that's 1-2 hue lighter as color_hl.

dbrgn, 2019-03-29:
Brushless controller project:

[image]

I wonder if we should reduce transparency for the copper layers a bit...

ubruhin, 2019-03-29:
> Here's the updated settings file, can you check?

Thanks! I would say the colors are good now :+1:

But I think the highlighted colors could be improved a lot by reducing transparency and maybe making the colors slightly more "aggressive". For example the non-highlighted vs. highlighted color of top copper are nearly the same, thus it's hard to see which traces are actually selected.

dbrgn, 2019-04-08:
New version: https://tmp.dbrgn.ch/settings.user.lp.draft4.txt

PR incoming in a minute.

PR #440 (2019-04-19): Update default layer colors
Still untested, I hope it compiles :slightly_smiling_face: 

Fixes #406.
closed 2019-04-19 by ubruhin