LibrePCB/LibrePCB #408: Focused Graphics Layer

shakenov-chinga, 2019-01-21:
Hi, everybody!
Made just a quick draft of focused layer feature. Please, take a look at [video](https://youtu.be/ICF90MCYlRw) first.
Mainly, this feature was intended to see if blind/buried via is on focused layer. You can't see it on the video, but via will be drawn gray, if it is not on that layer. Of course, there are a lot of cases that weren't taken into account.

There are, also, a lot of things that was difficult to do elegant way, so I implemented them straightforward ugly way:
1. I couldn't find any connection between BGI-item classes and `BoardEditor` class, so store focused graphics layer variable in `Board` class, and whenever focused layer changes I emit `attributesChanged` signal, which is leads to BGI-items' `paint` function. This is where I decide how draw the items.
2. For now, unfocused layers color (gray in a background) is stored as a static constant variable in `GraphicsLayer` class.
3. I weren't able to detect where polygon and text paint functions are implemented.

Now, if this feature is fit in this project, I ask your help to understand how layer focusing should work :smiley:. 
0. Airwires are not affected on by focused layers in any way, and always visible in Guide layer is not hidden.
1. Only one focused layer should be selected.
2. Draw Trace and Draw Plane commands focuses only copper layers.
3. If copper layer is selected, only that layer should be focused.
4. If layer that belongs to silkscreen is selected (ie one of Top/Bottom Name/Value/Place/.../etc), all silkscreen layers of that side and footprint of that side should be focused. It helps align only Texts and Polygons on that side only.
5. Highlighted items should always be seen. This is needed to be able to see whole signal while drawing the trace.
6. ...

I hope you like it. Best regards! :smile:

ubruhin, 2019-01-21:
Hi @shakenov-chinga,

> Made just a quick draft of focused layer feature. Please, take a look at video first. This is my development board I was doing as a hobby smiley

Wow, this is awesome! Both, the focused layer feature and your development board :smiley: 

> I couldn't find any connection between BGI-item classes and BoardEditor class, so store focused graphics layer variable in Board class, and whenever focused layer changes I emit attributesChanged signal, which is leads to BGI-items' paint function. This is where I decide how draw the items.

Yeah that's a bit ugly, but I also did something similar some time ago, see #67 ;) So generally I'm not against this "hack" until there is a nice solution for #67. But maybe you could at least move the "focused layer" attribute from `Board` to `BoardLayerStack`, as it's actually only related to the layer stack. And maybe you could emit the `GraphicsLayer`'s signal `attributesChanged()` instead of the signal from the board? That way it should only trigger the repaint of affected graphics layers, not from the whole board. But I'm not 100% sure if that will really work, it's just a quick idea...

> For now, unfocused layers color (gray in a background) is stored as a static constant variable in GraphicsLayer class.

Hmm could this color may be determined dynamically, for example by changing the saturation of the normal layer color by a specific value? Or should we give every layer a third color variant for their "disabled" state? Currently layers already have "normal" and "highlighted" colors, so it would somehow make sense to also add a "disabled" color or so...

> I weren't able to detect where polygon and text paint functions are implemented.

[`libs/librepcb/common/graphics/`](https://github.com/LibrePCB/LibrePCB/tree/master/libs/librepcb/common/graphics)

I'm looking forward for a pull request to test it by myself :)

shakenov-chinga, 2019-01-24:
Hi,
@ubruhin, I weren't able to find any connection between `PrimitivePathGraphicsItem` and `Board`. It can be hard-coded, passing `Board` instance via constructor, but all of my inner self is refusing to do so :smiley:.
I think it would be very wise to fix architecture issues before adding new features. Should I finish this feature any way possible, or maybe I can look what I can do to help with architecture fixes. I realize that it will take a lot of time and there are things in higher priority, but I think it will require just function/class/variable rearrangements, and code inside functions will not change in most cases. And maybe I am wrong :smile:.

Best regards!