LibrePCB/LibrePCB #410: Build release 0.1.0

ktrace, 2019-01-23:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.0
Git Revision:     d7458d3b3e126499902e1a66a0ef889f516a7c97
Build Date:       2019-01-22T16:44:05
Qt Version:       5.11.3 (built against 5.11.3)
CPU Architecture: x86_64
Operating System: Calculate/Linux
Platform Plugin:  xcb
```

##### SUMMARY

Build release from tarballs: appversion detected by git call, directory .git does not exist.

##### STEPS TO REPRODUCE

##### EXPECTED RESULTS

Releases must build without commit md5, for example, get version via variable.

##### ACTUAL RESULTS

Build broken and need patch.

##### ADDITIONAL INFORMATION

ubruhin, 2019-01-24:
I suspect with "tarballs" you mean the sources hosted at the GitHub releases page? If yes, those aren't functional, see release description. Please use the official source ZIP from [here](https://download.librepcb.org/releases/0.1.0/librepcb-0.1.0-source.zip) instead, which contains all submodules and doesn't need Git for compiling.

ktrace, 2019-01-24:
oh, sorry for noise, you are right, it's my fail. close this issue.

closed 2019-01-24 by ktrace