LibrePCB/LibrePCB #412: Device editor messages: Warn if no signals are connected

dbrgn, 2019-01-26:
When creating a new device, show a warning message if there are pads, but no pads have been connected to a component signal.

bcallebaut, 2019-07-18:
Can you elaborate ?
I did the following:
- I created a dummy symbol with 5 pins
- I created a dummy a package with 5 pins
- I created a Device based on the created symbol and package.
There is no warning/message if I let the pins unconnected in the messages box.
Is that what you want to add ?

dbrgn, 2019-07-19:
Hm, it's a while since I created this issue, but yes, I think that's what I meant:

[image]

Two pins, not connected to any component signal.

PR #680 (2020-04-04): Device editor: Show warning if no pads are connected
Shows a warning in the device editor if there are package pads, but none of them is connected. As soon as one pad gets connected, the message disappears.

I used the severity "warning" instead of "error" because there are situations where this message is a false-positive: If the package has only pads for mechanical reasons, but does not provide an electrical functionality, this message needs to be ignored.

Fixes #412
closed 2020-04-04 by ubruhin