LibrePCB/LibrePCB #432: Segfault when re-opening project

dbrgn, 2019-03-17:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.0-unstable
Git Revision:     bc82905bb1b9abbee9ac4fc79018122e07a2c1be
Build Date:       2019-03-17T16:37:56
Qt Version:       5.12.1 (built against 5.12.1)
CPU Architecture: x86_64
Operating System: Arch Linux
Platform Plugin:  xcb
```

##### SUMMARY

When closing and re-opening a project, on re-opening the project LibrePCB segfaults.

##### STEPS TO REPRODUCE

Possibly this has to do with the user settings file.

Try this:

    echo "(librepcb_board_user_settings)" > boards/default/settings.user.lp

Then open the project, close it again and re-open it.

ubruhin, 2019-03-18:
I can't reproduce this... Can you reproduce it with the binaries from https://download.librepcb.org/nightly_builds/master/? If not, it might be a local build issue which could be fixed by a clean rebuild ;)

dbrgn, 2019-03-19:
Happens with the nightly AppImage too.

ubruhin, 2019-03-19:
Okay, strange. Could you get a stacktrace? I tried it on two machines, no crash...

ubruhin, 2019-04-06:
Removed from milestone 0.1.1 since it is not really reproducible. Or can we even close this issue?

dbrgn, 2019-04-07:
I'll try to find the cause sometime, but yeah, I don't think it's a widespread problem.

Let's keep it open for a while to see if someone else experiences this problem.

alfishe, 2019-05-25:
I believe that there should be global error/exception handler for all load/save/export/import operations so any error will just roll back to previous state (or just open GUI without project loaded if it's app (re)start). Segfault is not an option for user-friendly app.

rnestler, 2019-06-17:
> I believe that there should be global error/exception handler for all load/save/export/import operations so any error will just roll back to previous state (or just open GUI without project loaded if it's app (re)start). 

Segfaults normally don't have anything to do with C++ exception handling, but most of the time are caused by memory safety bugs. While one could theoretically [set up a handler for the SIGSEGV signal](https://stackoverflow.com/questions/2350489/how-to-catch-segmentation-fault-in-linux) on Linux, it is almost impossible to do anything without triggering undefined behavior which may lead to much worse behavior than just aborting.
What one could do is to start multiple processes, where one supervises the other and handles the case when it aborts.

> Segfault is not an option for user-friendly app.

Of course these bugs should be fixed :slightly_smiling_face: and I agree that they should have high priority to get fixed.

PR #573 (2019-11-11): ControlPanel::projectEditorClosed: Use toUnique on file path
Because the `toUnique` call was missing, projects on a symlinked path
were not removed from the `mOpenProjectEditors` hash map. This in turn
would trigger a segfault when reopening a project due to a
use-after-free.

Fixes #432.
closed 2019-11-11 by ubruhin