LibrePCB/LibrePCB #436: Error list entries line wrapping

pschichtel, 2019-03-26:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.0
Git Revision: 2.21.0
Qt Version: 5.12.2
Operating System: Archlinux Stable
Flatpak: Yes
...
```

##### SUMMARY
The text in the error list entries is wrapping, but the items are not resizing. 

[image]

##### STEPS TO REPRODUCE
In my case:
1. Create a new project and place a bunch of bipolar capacitors
2. Default size of ERC widget forces the resulting "unconnected component" messages to wrap.

In the general case:
1. Open a project that has at least one error or trigger a new error
2. Drag the ERC widget narrower until the text starts wrapping

##### EXPECTED RESULTS
Either the list items should increase in size to show the full text properly or an ellipsis should be used to cut off text that doesn't fit.

##### ACTUAL RESULTS
Items that have line wrapping text are unreadable.

ubruhin, 2019-03-27:
Hi,

I'm not able to reproduce this, but I saw that the tree view has word wrap enabled, which actually doesn't work on my system with the official binaries (maybe it depends on the Qt version). I would propose to disable word wrap anyway as it probably looks very ugly with low widths, although I don't know if this would also fix this issue...

ubruhin, 2019-03-30:
OK now with Qt 5.12.2 I can reproduce it -> I will fix it quickly to have it in the 0.1.1 release.

PR #437 (2019-03-30): Schematic editor: Fix unreadable text in ERC dock with Qt 5.12
With Qt 5.12 the text in the ERC dock widget gets wrapped, but the height is not adjusted accordingly, thus the text is unreadable. Disabling word wrap fixes this.

Fixes #436
closed 2019-03-30 by ubruhin