LibrePCB/LibrePCB #445: File format: Let footprint pads have their own UUID and remove limitations

ubruhin, 2019-04-21:
##### SUMMARY

Currently the file format of footprint pads looks like that:

```scheme
(pad 40c1ac36-0911-4d9a-9109-8897d1953d94 (side top) (shape rect)
 (position 1.4 0.0) (rotation 0.0) (size 1.6 1.8) (drill 0.0)
)
```

where the UUID has two purposes:

- Unique identification of every footprint pad (required for the board editor).
- Connection to one of the package pads (the UUID is actually the package pad UUID, and the footprint pad doesn't have its own UUID).

This causes several limitations:

- A footprint pad cannot be unconnected. Unconnected footprint pads might be useful in some cases, e.g. for pure mechanical reasons, without electrical functionality.
- Due to the point above, copy&paste of footprint pads doesn't work if the pasted footprint pads can't be connected to package pads. This feels like a bug because pads are sometimes not pasted.
- A package pad can't be connected to multiple footprint pads. This would be needed for packages which have several solder points which are electrically connected, for example a shielding with multiple solder points.

##### SOLUTION

- [ ] Let footprint pads have their own UUID, independent of the package pad UUID:
    ```scheme
  (pad af3ead15-37a6-474d-8698-e4a22bd117de (side top) (shape rect)
     (position 1.4 0.0) (rotation 0.0) (size 1.6 1.8) (drill 0.0)
     (package_pad 40c1ac36-0911-4d9a-9109-8897d1953d94)
  )
    ```
- [ ] Allow unconnected footprint pads (`(package_pad none)` in the file format).
- [ ] Allow connecting multiple footprint pads to the same package pad.
- [ ] Remove the limitation of copy&paste in the package editor. Just leave pasted pads unconnected if no matching package pad is available.

rnestler, 2019-06-17:
I don't get why a footprint could have more pads than the package. I mean what is the use case for having more pads in the footprint than in the package? The package should always define all the possible pads (https://docs.librepcb.org/#libraryconventions-packages-pads) and the footprint may omit some of them, right?

ubruhin, 2019-06-26:
> I don't get why a footprint could have more pads than the package.

Some packages have pads for mechanical reasons only, without any electrical function. Example:

[image]

In these cases I think it doesn't make sense to add the mounting pads to the package pad list since it *never* makes sense to connect them electrically.

PR #807 (2020-11-26): Give footprint pads an UUID and allow unconnected pads
Let footprint pads have their own UUID instead of using the UUID of the connected package pad.

When upgrading v0.1 package files to v0.2, the key `package_pad` gets added. Then the `pad` and `package_pad` both have the same value:

```scheme
(pad af3ead15-37a6-474d-8698-e4a22bd117de (side top) (shape rect)
 (position 1.4 0.0) (rotation 0.0) (size 1.6 1.8) (drill 0.0)
 (package_pad af3ead15-37a6-474d-8698-e4a22bd117de)
)
```
This way the file format upgrade is deterministic. When creating entirely new packages in LibrePCB v0.2.x, the footp [... cut]
closed 2020-11-26 by ubruhin