LibrePCB/LibrePCB #450: SchematicEditor: Mirror action produces offsets for the whole selected area

alfishe, 2019-05-12:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.1-unstable
Git Revision: c14084c441
Qt Version: 5.12.0
Operating System: Windows 10 (10.0.17763 Build 17763)
...
```

##### SUMMARY
SchematicEditor: Mirror action produces offsets for the whole selected area. Offset errors are accumulated if multiple mirroring operations performed

##### STEPS TO REPRODUCE
1. Open "Demo Brushless Controller project"
2. Open schematic editor
3. Select rectangular area covering R29, R30, C22 and V_BATTERY net area
4. Press 'M' or perform mirror from Edit menu multiple times

##### EXPECTED RESULTS
On even number of sequential mirroring operations schematic fragment stays on same position with no shifts / offsets
[image]

##### ACTUAL RESULTS
On each mirror - fragment shifts left. Offset error accumulates on each mirror operation triggered
[image]

##### ADDITIONAL INFORMATION
Seems related to mirror area center coordinates specifics. Instead of using boundary rectangle, mirroring command recalculates center based on items coordinates. Doesn't look and feel right

ubruhin, 2019-05-13:
This is strange indeed ;) 

I think the same issue exists in the board editor, and probably also in symbol and package editors.

ubruhin, 2024-11-30:
I can't reproduce this anymore with the current `master` version, probably this has been resolved in the mean time. Thus closing.

closed 2024-11-30 by ubruhin