LibrePCB/LibrePCB #452: got errors  compile file"optional.hpp"  on win7 &qt5.2 &mingw4.8

cailei0596, 2019-05-18:
some errors :

  TL_OPTIONAL_11_CONSTEXPR T &get() & { return this->m_value; }
  TL_OPTIONAL_11_CONSTEXPR const T &get() const & { return this->m_value; }
  TL_OPTIONAL_11_CONSTEXPR T &&get() && { return std::move(this->m_value); }

E:\MyWork\pcb\librepcb-0.1.1\libs\optional\tl\optional.hpp:434: error: expected ';' at end of member declaration
   TL_OPTIONAL_11_CONSTEXPR T &get() & { return this->m_value; }
                                   ^
E:\MyWork\pcb\librepcb-0.1.1\libs\optional\tl\optional.hpp:434: error: expected unqualified-id before '{' token
   TL_OPTIONAL_11_CONSTEXPR T &get() & { return this->m_value; }
                                       ^
anyone has encountered this problem?

ubruhin, 2019-05-19:
Hmm what compiler version do you have exactly? I just noticed that 4.8.1 seems to be the minimum required version to compile `tl::optional`. If you have >= 4.8.1, could you check if it compiles after replacing the file `libs/optional/tl/optional.hpp` with the version from [here](https://github.com/TartanLlama/optional/blob/master/tl/optional.hpp)?

ubruhin, 2019-11-30:
Closing this because of missing activity and related to an outdated version of LibrePCB. Feel free to reopen if the problem still exists.

closed 2019-11-30 by ubruhin