LibrePCB/LibrePCB #454: Feature Request

ajpcode, 2019-05-24:
##### SUMMARY

Firstly, thank you for your work on this project, I've just completed and sent off my first design for manufacture using this software.

This feature request relates to #196 "Add support for copper planes in boards".

Whilst working on a recent design I thought It might be nice to have:
- Thermal spokes on all pads (not vias) connected to/located within a ground plane.
- The ability to connect traces to ground planes (although my requirement of this might be satisfied if the first request was met as the plane could fill the layer). During my recent design I was forced to place ground vias to connect to the plane.

##### SOLUTION
Maybe a copper-pour keep-out region similar to non-pour nets around the pads with four equidistant spokes connecting out to the matched plane.

dbrgn, 2019-05-24:
> I've just completed and sent off my first design for manufacture using this software.

That's awesome to hear, would be great if you could post some pictures at https://librepcb.discourse.group/t/projects-madewithlibrepcb/99 when you receive the resulting PCBs :slightly_smiling_face:

ubruhin, 2019-05-27:
> Thermal spokes on all pads (not vias) connected to/located within a ground plane.

Yes, this definitively needs to be added. Unfortunately it's not trivial to calculate the spokes, there are multiple possible solutions with different advantages/disadvantages and I'm not sure which one is the best. Would be nice if someone could dive into this to find the best option ;)

> The ability to connect traces to ground planes

Hmm yeah honestly I'm not sure why I didn't implement this. Probably I though that where a trace is, there is also a via or pad connected to, but actually that's not always true :thinking:

ajpcode, 2019-07-14:
> That's awesome to hear, would be great if you could post some pictures at https://librepcb.discourse.group/t/projects-madewithlibrepcb/99 when you receive the resulting PCBs slightly_smiling_face

Apologies, photos aren't great but... uploaded.

dbrgn, 2019-10-19:
> The ability to connect traces to ground planes

I think this has been implemented in the meantime!

EternityForest, 2020-06-25:
What about just adding a very basic initial implementation, where you take the center point of any pad, and draw four lines outwards, marking those regions as exempt from the keepout logic for that particular hole?

Or even simpler,  generate two copper plane maps, one with no isolation, and one with.   The difference between them is the region subject to adding thermals, find the center points of pads, draw spokes outwards, mask those lines based on that difference, then merge back into the plane.

brim2full, 2021-01-16:
> Yes, this definitively needs to be added. Unfortunately it's not trivial to calculate the spokes, there are multiple possible solutions with different advantages/disadvantages and I'm not sure which one is the best. Would be nice if someone could dive into this to find the best option ;)
<

I'm not convinced it is necessary to automatically calculate the spokes and their orientation etc.  Automatic calculation is not done for traces or vias.  The DRC enforces the minimum standards and it up to the pcb designer to talk to the circuit engineer and determine the current (A) and inductance (H) parameters to be acceptable.  Or not.

I think its only really necessary to have some project design parameters set: pad to plane clearance, number of spokes and spoke thickness.  Then allow these to be optimised on a pad by pad basis.  Once set the clearance and traces are automatically drawn.  The traces are not fixed but can be repositioned as required.

dbrgn, 2021-01-19:
There are some more thoughts by @brim2full in this thread: https://librepcb.discourse.group/t/how-to-create-a-thermally-relieved-connection-to-a-plane/347

doom-fr, 2021-06-24:
I would say easier algorithm is this one : 
- calculate the plane as all pads were not of plane net
- for each plane net pad : 
   - calculate spokes_width = max( (pad_width+pad_height) / 16 , min_copper_width , plane_min_width)
   - for each spoke's direction (up,down,left,right) : 
      - start from centre and go in this spoke's direction with a length of (pad_width_or_height/2+plane_clearance) and a width of spokes_width
      - if it does not respect clearance with other net => remove the spoke, else keep it

DRC will help to see if all goes well.

doom-fr, 2021-06-24:
Other option : 

- calculate the plane as today
- for each plane net pad : 
    - calculate spokes_width = max( (pad_width+pad_height) / 16 , min_copper_width , plane_min_width)
    - calculate pad_shape extended of clearance - centred_horizontal_line - centred_vertical_line
    - subtract this from the plane

doom-fr, 2021-06-24:
[image]

Purple = red - green - blue - yellow

[image]

Then subtract purple from the calculated plane

doom-fr, 2021-07-21:
Here is a bad workaround to wait the functionality : 

[image]

It is really far from perfection and produces a lot of DRC errors that  need to be checked manually.

ubruhin, 2021-07-21:
Thanks @doom-fr for the suggestions above, I'll check them in detail once I start implementing this feature.

Regarding the workaround, there is also an easier workaround available which does not produce any DRC warnings:
- Plane properties: Set the plane connect style to "None"
- Plane properties: Check "Keep Orphans" (optional, just avoids temporarily disappearing plane if no traces are added yet)
- Add a short trace to each pad to be connected to the plane

[image]

doom-fr, 2021-07-22:
Thanks for this better workaround.
It needs some work (4 layers & many through hole connectors), but it works great.

[image]

I separated the plane in 2 parts, one with 'none' and one with 'solide' as I have a lot of components and I think it is better for SMD pads, especially SMD thermal pads.

Well rendered with Ucamco gerber viewer :
[image]

doom-fr, 2021-07-22:
I think it would great to have an 'hybrid' mode : 'solide' for everything except for THT pad which will be 'none'

gx-turbo, 2022-12-30:
I just started my first small design for test. I guess I will make an order at AISLER during this weekend.
And these thermal spokes are up to now really the only thing I miss. For the small test-design the workarounds above are good enough. But in general, it is a must-have.
congratulations and thank you for the good work!

ubruhin, 2023-01-06:
Are there any opinions here about thermal reliefs on **vias**? Currently LibrePCB respects the "connect style" plane property also for vias. So if you choose "None", vias are not connected to the plane at all. And once we support thermal reliefs, vias will also be connected with thermal reliefs if that connect style is chosen.

But I wonder if this really makes sense. Vias are not soldered, so heat dissipation is not an issue - actually one often even *want* to have good heat transfer from one copper layer to the others, while heat transfer on pads on the same plane is not desired.

So I'd say we have two options:
1. Implement two independent "connect style" properties - one for vias, one for pads. This would allow maximum flexibility and a 100% backwards compatible file format upgrade, but also adds complexity which might be completely unnecessary.
2. Do not respect the "connect style" property for vias at all. Instead, always connect them solid. It seems that's also the way KiCad implemented it. A downside is that it may modify planes during the 0.1->0.2 file format upgrade, since unconnected vias will be connected afterwards. However, I think that's not a big problem (no risk for shorts since clearances are respected of course, just some traces might be pointless afterwards if they were added to emulate thermal spokes as suggested in the workaround above).

Currently I tend to go with variant 2., also because KiCad doesn't have a more flexible option IMHO it's a good sign that the flexibility of option 1. is really not needed :wink:

Another similar question is about SMT pads vs. THT pads. Here KiCad does make a difference as it allows to apply thermal reliefs only to THT pads while connecting SMT pads solid. I'm not sure whether this is really a useful feature or not...

gx-turbo, 2023-01-06:
from my point of view, I agree to option 2., because Vias are not supposed to be soldered, they should have solid connections. For special cases you can always create a dummy part with one THT pad. 
SMT pads I believe should have thermal reliefs, although you can find some sophisticated thoughts (which may apply mostly to multilayer PCBs) to that topic, see below.

https://www.tempoautomation.com/blog/how-to-design-thermal-reliefs-with-minimal-manufacturing-issues/
https://www.ourpcb.com/thermal-pad-pcb.html
https://resources.pcb.cadence.com/blog/2021-pcb-thermal-relief-guidelines-for-effective-layouts
https://resources.altium.com/p/thermal-relief-design

Tubbles, 2023-02-13:
> Are there any opinions here about thermal reliefs on **vias**?

I'd say go with option 2. If kicad does it like that then its indeed a very good indication that option 1 is really not needed.

Although personally I do feel like it should be overridable on a per pad basis, and if that's implemented then I think throwing vias in there as well might make sense. But that probably warrants a new ticket :)

ubruhin, 2023-02-13:
OK thanks for the feedback! I have now implemented this change for vias in #1100.

ubruhin, 2023-05-05:
FYI, I think I have found a reasonable algorithm to calculate thermal spokes, see explanation in #1160. If somebody finds a problem with this algorithm, please let me know :slightly_smiling_face:

gx-turbo, 2023-05-05:
GREAT!

closed 2023-05-07 by ubruhin