LibrePCB/LibrePCB #459: Build fails on delauney-triangulation

luigino, 2019-06-02:
Using 0.1.1 zip from main site with openSUSE Tumbleweed and gcc9.
This is what happens almost immediately:
```
cd delaunay-triangulation/ && ( test -e Makefile || /usr/lib64/qt5/bin/qmake -o Makefile /home/abuild/rpmbuild/BUILD/librepcb-0.1.1/libs/delaunay-triangulation/delaunay-triangulation.pro 'QMAKE_CXXFLAGS= -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -g -DOPENSSL_LOAD_CONF' 'QMAKE_CFLAGS= -O2 -Wall -D_FORTIFY_SOURCE=2 -fstack-protector-strong -funwind-tables -fasynchronous-unwind-tables -fstack-clash-protection -g -DOPENSSL_LOAD_CONF' 'QMAKE_LFLAGS= -Wl,--as-needed -Wl,--no-undefined' ) && make -f Makefile 
make[2]: Entering directory '/home/abuild/rpmbuild/BUILD/librepcb-0.1.1/libs/delaunay-triangulation'
mkdir -p build
g++ -o build/main.o -c main.cpp -std=c++11 -Wall -Wextra -Wfatal-errors -pedantic -w -Winline -fno-rtti -ggdb -D_GLIBCXX_DEBUG -lsfml-graphics -lsfml-window -lsfml-system
main.cpp: In function 'int main()':
main.cpp:30:14: error: 'Vector2' was not declared in this scope
   30 |  std::vector<Vector2<float>> points;
      |              ^~~~~~~
compilation terminated due to -Wfatal-errors.
```

ubruhin, 2019-06-02:
What commands have you used to compile? Actually the `main.cpp` in delaunay-triangulation shouldn't be compiled at all (it's not used for LibrePCB) when following [our build instructions](https://github.com/LibrePCB/LibrePCB#using-qmake-and-make).

luigino, 2019-06-02:
Oh, that -r switch fixed it. My bad.

closed 2019-06-02 by luigino