LibrePCB/LibrePCB #463: The GitHub Sponsor botton shows an error

rnestler, 2019-06-07:
##### SUMMARY
Clicking the GitHub Sponsor button shows an error:
[image]

##### STEPS TO REPRODUCE
Click the GitHub Sponsor button.

##### EXPECTED RESULTS
Showing how to sponsor the LibrePCB developers

##### ACTUAL RESULTS
The error message

##### ADDITIONAL INFORMATION
I think @ubruhin needs to join the Sponsors waitlist for the error to disapear: https://github.com/sponsors
[image]

ubruhin, 2019-06-07:
> I think @ubruhin needs to join the Sponsors waitlist for the error to disapear

I did it already some time ago, but I can do nothing else than waiting to get accepted ;)

ubruhin, 2019-10-21:
Fixed :)

closed 2019-10-21 by ubruhin