LibrePCB/LibrePCB #466: Supply Net Label renames don't get propagated to wires

rnestler, 2019-06-15:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.1-unstable
Git Revision:     e8586f91a4f68af7245c0d45e45f2e8bce5d8a66
Build Date:       2019-06-15T01:00:31
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Arch Linux
Platform Plugin:  xcb
```

##### SUMMARY
When renaming a Supply Net label, the rename doesn't get propagated to the connected wires, resulting in a ERC error.

##### STEPS TO REPRODUCE
Create a supply net label and connect it
[image]
Rename the supply net label.
[image]

##### EXPECTED RESULTS
The net label of the wire should also change

##### ACTUAL RESULTS
The net label of the wire doesn't change, resulting in an ERC error.

5n8ke, 2020-09-03:
To implement a solution some aspects need to be clarified.

1. The supply symbol allows values, which are then used for the net name, to contain spaces and even newlines. This is not allowed for Netlabels, which also does not allow other special characters (Umlaute, asterisks, ...). Should the allowed values be restricted or the netlabel restrictions be lifted?
2. Should this also change the value of all other supply symbols, that are connected to the segment?
3. What about symbols that force a net, but don't use their variable value, but a fixed name?
4. When changing the value, should the question be asked, whether to change the whole signal or just the 1 segment?

ubruhin, 2020-09-26:
> The supply symbol allows values, which are then used for the net name, to contain spaces and even newlines. This is not allowed for Netlabels, which also does not allow other special characters [...]

Currently an error message appears when using illegal net names, and I'd simply keep that behavior for now. I think the problem described in this issue can be solved without changing the restrictions on net names.

> Should this also change the value of all other supply symbols, that are connected to the segment?

This is almost impossible with the current concept. Component signals (defined in the library editor) can either be static, or dynamic with attribute placeholders (usually `{{VALUE}}`). Static net names can't (and shouldn't) be changed at all, and dynamic net names would be way too complex (for example, the value of a component could contain placeholders too).

We *could* implement it for the case where the forced net name is `{{VALUE}}` and the value of a component is a fixed string. But generally I don't like such workarounds for very specific cases (i.e. this solution would not be generic).

Maybe we should simply rename the net if a supply symbol value was modified, no matter whether there are other supply symbols connected to the same net? Most net segments don't have more than one supply symbol anyway, in these cases it will do exactly what the user expects. And if other supply symbols are connected, the ERC shows a warning about the inconsistent net name, so it's not that dangerous.

> When changing the value, should the question be asked, whether to change the whole signal or just the 1 segment?

Yeah, that would be nice!

A much more fundamental question from my side: Are there any ideas how to implement supply symbols in a better way? I'm not sure if the current concept of forced net names in component signals is the best way, but I didn't find a better solution yet. And for sure I don't want to have separate components for each net supply (VCC, VDD, 3V3, ...) like other EDA tools....