LibrePCB/LibrePCB #473: Gerber output: Filled polygons

dbrgn, 2019-07-01:
### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: 0.1.1-unstable
Git Revision:     e8586f91a4f68af7245c0d45e45f2e8bce5d8a66
Build Date:       2019-06-14T09:01:51
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Arch Linux
Platform Plugin:  xcb
```

##### SUMMARY

A filled polygon on the top copper layer is not properly outputted in the gerber output.

##### STEPS TO REPRODUCE

Create a polygon on the top copper layer in the board editor. Export to Gerber. Open with gerbv.

[image]

##### EXPECTED RESULTS

Filled polygons.

##### ACTUAL RESULTS

Polygons are not closed.

[image]

dbrgn, 2019-07-01:
Ah, I just realized that the LibrePCB polygons are not properly closed either. But even when a polygon is closed, it's not filled in the Gerber output.

So:

- Warn if filled polygons are not closed when exporting to Gerber
- Actually fill polygons when exporting to Gerber

ubruhin, 2019-07-07:
> Warn if filled polygons are not closed when exporting to Gerber

Instead of a warning, I would prefer that the board editor and the Gerber output always look the same. There are two different ways to achieve this:

1. Either fill the polygons in the Gerber output, the same way as the board editor does.
2. Or don't fill non-closed polygons in the board editor, even if the "closed" property of the polygon is set.

Variant 1 would be more flexible since it allows to draw such polygons:

[image]

But I think such polygons are not really a use-case and could even be "dangerous" (with thinner lines, the user might not realize that the polygon is not properly closed and thus the outline is not as intended). So I would say variant 2 makes more sense. Do you agree @dbrgn?

dbrgn, 2019-07-07:
Yeah, I think 2 makes sense! Maybe it could be filled while drawing/editing, but not anymore once you exit the drawing mode.

PR #479 (2019-07-08): Fix inconsistent appearance of polygons between editor and Gerber export
- Only fill polygons if they are closed (filled but non-closed polygons look strange -> avoid them).
- Fix missing area of filled polygons in Gerber export (filled polygons were drawn correctly in editors, but wrong in Gerber export).

Fixes #473 

#478 must be merged first since CI is currently broken for MacOS.
closed 2019-07-08 by ubruhin