LibrePCB/LibrePCB #474: Scrolling with Mac trackpad zooms, but should scroll

jeffwheeler, 2019-07-03:
##### SUMMARY

Presently, the `GraphicsView` in the schematic and layout editor is difficult to use with a Mac trackpad. When swiping north with two fingers, the widget zooms out, and when swiping south with two fingers, the widget zooms in. Pinching does not work as it normally does on Mac OS.

Expected behavior can be seen in most apps, such as Preview.app.

##### SOLUTION

Standard Mac OS behavior is to scroll in two dimensions using two fingers, and pinch to zoom in/out. Qt supports this with [`QNativeGestureEvent`](https://doc.qt.io/qt-5/qnativegestureevent.html), but after playing with it for a bit I couldn't get it to work quite right.

(Side note, I found [this blog post](https://blog.qt.io/blog/2017/01/19/should-you-be-using-qgraphicsview/) warning that `QGraphicsView` may be deprecated in the future.)

jeffwheeler, 2019-07-03:
I've posted [this commit](https://github.com/jeffwheeler/LibrePCB/commit/8573842955dc68f7dea7d0b4f47a532ab2d9a64e) which demonstrates using `QNativeGestureEvent`. It works, but I expect it breaks other systems (and Macs with mice) because I disable the scrollwheel handling.

ubruhin, 2019-07-03:
Thanks for the code example! Looks like it shouldn't be too hard to integrate that code snippet properly. @jeffwheeler would you be interested to create a pull request for that?

closed 2019-07-25 by jeffwheeler