LibrePCB/LibrePCB #475: Connection "required/optional" checkbox

mwit-a, 2019-07-03:
##### SUMMARY
In my opinion, the 'connection required' column can be improved. 
I think it is not intuitive if you need to check a checkbox with the label 'optional' to change the status to 'required'.

##### SOLUTION

Rename the column into 'Connection required' and just leave the checkboxes.

[image]

ubruhin, 2019-07-03:
To be honest, the main reason why I did it that way is that in #456 I switched to the native checkbox support of `QTableView` which unfortunately does not allow to center the checkbox inside its cell, and the left aligned checkbox IMHO looks ugly. So I just added the text "Optional" resp. "Required" to the cell to make it looking better :grin: Actually I expected that this should even be more intuitive than a checkbox without text...

rnestler, 2019-07-04:
> Actually I expected that this should even be more intuitive than a checkbox without text...

If one sees
 * [ ] Optional

one may assume that it is not Optional since optional is not checked.

dbrgn, 2019-07-04:
I agree with that, it's a bit confusing :slightly_smiling_face:

dbrgn, 2020-08-17:
Two potential solutions:

- A: Remove the checkbox label, but change the column label from "Connection" to "Required"
- B: Change the checkbox label to a static "Required" text

@ubruhin would you be happy with one of those two?

ubruhin, 2020-08-17:
> @ubruhin would you be happy with one of those two?

- A: Would be fine for me if the checkbox is then centered, but I'm pretty sure it would be cumbersome to implement ;)
- B: Not as good as A, but probably still better than the current solution - so yes, would be fine for me :)

PR #875 (2021-03-05): Component editor: Remove checkbox text "required/optional"
Removed the text of the component signal checkbox for choosing "required" or "optional" to avoid confusion what state the checkbox actually has. Instead, the column title "Connection" is now changed to "Required" and the tooltip is more descriptive.

[image]

It's a bit ugly now since the checkbox is not centered. I tried to center it but found no reliable, simple way to do so...

Fixes #475
closed 2021-03-05 by ubruhin