LibrePCB/LibrePCB #483: Improve Polygon Editor

dbrgn, 2019-07-13:
##### SUMMARY

The polygon editor is currently very limited. The biggest problem ist that rows can't be moved.

For example, here:

[image]

...it's not easily possible to add more points between row 1 and 2.

##### SOLUTION

- Allow moving rows around
- Allow inserting rows below a selected other row
- Maybe allow copying rows

closed 2019-07-30 by ubruhin

EdizonTN, 2019-09-08:
Notice!
I try nightbuild and I found small problem:
Entering value into pos.X/Y of polygon segment in polygon properties window, is limited only 2 decimal places! Please, check it.

LibrePCB Version: 0.1.3-unstable
Git Revision:     466ea012804a28243fea50d877329a1c89cff945
Build Date:       2019-09-02T19:47:06
Qt Version:       5.11.3 (built against 5.11.3)
CPU Architecture: x86_64
Operating System: Windows 7 SP 1 (6.1)
Platform Plugin:  windows

ubruhin, 2019-09-08:
Thanks for the report @EdizonTN, I opened #512 to track this issue.

PR #489 (2019-07-30): Polygon editors: Support copying and moving rows
Adding buttons for copying and moving rows in the polygon editors:

[image]

In addition, cells now show the unit of the contained value (`mm` and `°`).

Fixes #483