LibrePCB/LibrePCB #493: Circular elements do not appear to have rotational transformation applied in Gerber output

ajpcode, 2019-08-06:
##### VERSION / OS / ENVIRONMENT

```
LibrePCB Version: **0.1.1-unstable**
Git Revision: b73e8c97606da76d302b869a8de0c4b7231bc7ba (master)
Qt Version: 5.12.2
Operating System: Ubuntu Linux
...
```

##### SUMMARY

On generation of gerber fabrication data circular silkscreen elements do not appear to have rotational transformations applied to them that have been applied to the rest of the footprint elements. The LibraPCB board viewer shows the rotation correctly applied but on generation of gerber/fabrication files the rotation has not been applied to the circular elements.

This issue also applies to v0.1.0.

##### STEPS TO REPRODUCE

Create a new library footprint that contains an off-origin circle that will be used as part of a device (and assign to device).
Add the device in schematic editor.
Place the device in board view with a rotation e.g. 180 deg.
File > generate fabrication data > Generate Gerber & Excellon Files (I'm using 'protel naming').
Open the gerber files in gerbv/other to see that the rotation has not been applied to circular elements in the gerber output. (This is not isolated to gerbv - my last design had pin markings in the wrong places).

##### EXPECTED RESULTS

I would have expected the a circular element (for example denoting pin 1) on the bottom left of a footprint to be outputted to the top right after rotating through 180 degrees in placement.

##### ACTUAL RESULTS

No rotation is applied on output of fabrication data (but is displayed correctly in LibraPCB's board viewer). Despite rotating a component through 180 degrees the circular elements will remain in their original footprint position (e.g. in the bottom left instead of the top right)

##### ADDITIONAL INFORMATION

I can take screenshots if required.

Kind regards.

dbrgn, 2019-08-07:
@ubruhin how can the reported LibrePCB version be "0.1.1-unstable" if the referenced commit is up to date? maybe you forgot to update the version string somewhere?

Anyways, I can confirm this behavior!

ubruhin, 2019-08-09:
Thanks for the report! Looks like the issue is located in `BoardGerberExport::drawFootprint()` where footprint rotation is not handled...

> @ubruhin how can the reported LibrePCB version be "0.1.1-unstable" if the referenced commit is up to date? maybe you forgot to update the version string somewhere?

Yeah, sometimes I forget to bump the version on `master`... Will fix it and update the release checklist :)

dbrgn, 2019-08-21:
Argh, this just bit me, when I soldered in some diodes the wrong way around... :man_facepalming: :sweat_smile:

PR #508 (2019-09-01): Gerber export: Fix possibly wrong position of circles in footprints
Fixes #493.
closed 2019-09-01 by ubruhin