LibrePCB/LibrePCB #494: Change footprint on existing device caused crash

EdizonTN, 2019-08-08:
##### VERSION / OS / ENVIRONMENT
LibrePCB Version: 0.1.1-unstable
Git Revision:     87f729f41f28faeea5884a76a0a54516b58fe148
Build Date:       2019-07-19T13:39:29
Qt Version:       5.12.0 (built against 5.12.0)
CPU Architecture: x86_64
Operating System: Windows 7 SP 1 (6.1)
Platform Plugin:  windows

##### SUMMARY
After change of footprint on existing device in local library, casuses program crash.

##### STEPS TO REPRODUCE
- Open local library
- select device (ex. chip capacitor) and open it for edit
- click button with small magnifier (select another footprint)
- select another fooprint (ex. chip Resistor) with same pincount and pin names and click OK
- program crash

##### EXPECTED RESULTS
Change footprint

##### ACTUAL RESULTS
Program crash.

##### ADDITIONAL INFORMATION
Application output from QT:
_[  FATAL  ] ASSERT: "last < rowCount(parent)" in file itemmodels\qabstractitemmodel.cpp, line 2783 (itemmodels\qabstractitemmodel.cpp:2783)
Invalid parameter passed to C runtime function.
Invalid parameter passed to C runtime function._

[image]

ubruhin, 2019-08-09:
I can't reproduce this, even with the nightly builds from https://download.librepcb.org/nightly_builds/master/.

If you built LibrePCB by yourself, have you already tried a clean rebuild? Somethings there are strange build issues causing such crashes, and after a clean rebuild everything works fine.

EdizonTN, 2019-08-10:
Ok, forgot for this. Night build works fine. Problem is in my Qt maybe.

closed 2019-08-10 by EdizonTN

rnestler, 2019-09-05:
@EdizonTN It would be interesting to know which Qt version you had locally, which caused the crash.

EdizonTN, 2019-09-05:
@rnestler, it was builded from master from github at: "Build Date: 2019-07-19T13:39:29".

rnestler, 2019-09-09:
> I was wrote in first post :-)

@EdizonTN Oh I somehow missed this :see_no_evil:, sorry.