LibrePCB/LibrePCB #514: Library editor: Package edit - add an ability to show/hide layers

EdizonTN, 2019-09-08:
##### SUMMARY
In package editor (in library editor) please add an layer widget as in PCB editor.
Switching layers Visible/Hide, helps to check designed package (silkscreen view, documentation view,...)

##### SOLUTION
Add a layer widget as is PCB Editor (see attached picture)

ubruhin, 2019-09-08:
See also [this discussion](https://librepcb.discourse.group/t/package-editor-layer-filtering/108).

EdizonTN, 2024-04-04:
Any progress?
Switching layers On or Off VERY helps to check designed package (silkscreen view, documentation view,...) and selecting overlapping objects!
Pls....

ubruhin, 2024-04-04:
No progress yet, sorry... But I am aware that this would be helpful.

kagehisa, 2024-04-28:
I just stumbled upon this issue due to the [discussion](https://librepcb.discourse.group/t/package-editor-layer-filtering/108/5). I also would like to see this feature added. I would go so far as to say it is not just useful but neccessary to increase the usability of the Package Editor.
Creating a Package with drawings on all the different layers is almost impossible to alter atm.