LibrePCB/LibrePCB #515: Schematic editor: Add move and rotate functionality, for Name and Value fields

EdizonTN, 2019-09-08:
##### SUMMARY
In schematic editor will be beautiful, if component's fields "Name" and "Value" can be movable and rotateable.

##### SOLUTION
Example:
Default components place into schematic:
[image]

After Name and Value move and rotate:
[image]

doom-fr, 2019-10-13:
Yes, it would be useful to save space ...

[image]

EdizonTN, 2020-12-14:
Hello ubruhin,
Is it possible to increase priority of this feature request? Is able to implement in next version (0.1.6)?
Thanx.

PR #1085 (2023-01-24): Schematic: Detach symbol texts to make them editable
Just like in the board editor, this makes symbol texts (e.g. `{{NAME}}` and `{{VALUE}}`) movable, editable and can even be removed. In addition, there's now also a "Reset All Texts" (context) menu item which restores the original text properties from the library symbols.

Affects the file format of schematics since the text objects are now copied from the symbols into the schematics.

Closes #515
closed 2023-01-24 by ubruhin