LibrePCB/LibrePCB #517: Replacing component in schematic

EdizonTN, 2019-09-15:
in case https://librepcb.discourse.group/t/replacing-component-in-schematic/132 ....

##### SUMMARY
In schematic editor is absent option to change placed and connected component. One way is delete component and place a new one. But this way clear existing wire from/to previous deleted component.
Better way will be add a possibility to change connected components via new item in floating menu, without delete connected wires.

Is able to change components with different count of pins? (ex, change connected microcontroler with another one - standard situation)
I think yes, place it to same coord and designer must check all connected  and unconnected pins.

##### SOLUTION
Right click to selected component -> select "Change component" from floating menu -> select new one and click OK. That's  all.

Consequence in PCB:
If relevant footprint (device) is also plased on PCB and is routed, I propose to clear oldest device from PCB (but not tracks!) and new device show in "Place devices" dock window.

doom-fr, 2019-10-13:
For now, is it possible to just not delete wires in schematic (and board) when we delete a device ?
And if the actual functionality is needed, add it in the context menu.

ubruhin, 2019-10-14:
> For now, is it possible to just not delete wires in schematic (and board) when we delete a device ?

No, that's not possible yet...