LibrePCB/LibrePCB #526: Library Editor: Bad substitutional strings in textComboBox

EdizonTN, 2019-09-28:
##### VERSION / OS / ENVIRONMENT

LibrePCB Version: 0.1.2
Git Revision:     acdd94d9d2310f79215125b999153e9da88a9376
Build Date:       2019-07-13T21:10:26
Qt Version:       5.11.3 (built against 5.11.3)
CPU Architecture: x86_64
Operating System: Windows 7 SP 1 (6.1)
Platform Plugin:  windows

##### SUMMARY

I try to create sheet frame component and found mistake. When I try put {{..}} string into my components, List of this (substitutional?) special string in text combobox are in bad format:
[image]

After placing this components into schematic, strings not shown (not converted to their values).

I create a test component and result is here (left are typed string and right are their value):
[image]

Strings begining with "_" character, not working. Please, clear it.

##### ADDITIONAL INFORMATION
Strings are in:
...\LibrePCB\libs\librepcb\libraryeditor\sym\fsm\symboleditorstate_drawtextbase.cpp, line 92 to 97
and
...\LibrePCB\libs\librepcb\libraryeditor\pkg\fsm\packageeditorstate_drawtextbase.cpp, line 92 to 96

ubruhin, 2019-10-01:
Hmm that's indeed a leftover which is no longer valid...

Although I would generally not recommend to use built-in placeholders like `{{PROJECT}}` directly, I think it makes sense to remove all the underscores from the combobox. The reason why I don't recommend it is that when using a built-in attribute, the user has no way to override the content in a schematic. For example if the user wants to print a custom attribute (e.g. `{{COMPANY_NAME}}`) instead of `{{PROJECT}}`, he would need to create his own schematic frame since the frame in the Base library has `{{PROJECT}}` "hardcoded".

Thus the schematic frame in the Base library uses the placeholder `{{FIELD_PROJECT}}` instead, and the corresponding component contains an attribute which maps `FIELD_PROJECT` to `{{ SCH_FRAME_PROJECT or PROJECT }}`. This allows the user to set e.g. `SCH_FRAME_PROJECT` to a custom value (e.g. company name) without the need to create a new symbol/component.

Anyway, I think it makes sense to just remove the underscores from the combobox :) The concept with wrapped attributes is kind of "advanced" and difficult to provide in the GUI...

mexchip, 2019-10-18:
I'd like to help with this, what I'd like to know is if removing the underscore {{_ values is enough or they should be replaced by non underscore ones.

ubruhin, 2019-10-19:
> I'd like to help with this, what I'd like to know is if removing the underscore {{_ values is enough or they should be replaced by non underscore ones.

I think it makes more sense to only remove the underscores - without underscore, those placeholders work properly.

PR #555 (2019-10-20): Library editor: Remove underscore from placeholders
Fixes #526
closed 2019-10-20 by ubruhin