LibrePCB/LibrePCB #527: Board editor : add device by double-click

doom-fr, 2019-10-02:
##### SUMMARY
In pcb, to place a device we have to click on it and go to the bottom of the window to click "add" button

##### SOLUTION
Double-click on the device starts the adding process !
It is simple to do (I think) and user friendly !

ubruhin, 2019-10-05:
The only question is which device and footprint to choose then, since the user wouldn't be able to choose. Maybe allow the double-click only if the device is already pre-selected in the schematic (see also #529) to avoid adding a random (probably wrong) device?

The footprint is less critical, we should then just use the default footprint of the selected package.

doom-fr, 2019-10-06:
Maybe just use what is written in the device and footprint fields. If the user needs another one, he will change before.
And also allow to drag and drop the footprint representation (as it is a very big field, it will be the easiest and fastest method with the mouse).

I also noticed that if I change the device (shouldn't it be device type ?) and/or footprint of a component, click on another one and go back to the first, the changes are lost.

ubruhin, 2021-06-19:
With #892, devices can now be added with double-click so I think we can close this :tada:

closed 2021-06-19 by ubruhin