LibrePCB/LibrePCB #529: Schematic Editor : show and change device / package (keep the symbol)

doom-fr, 2019-10-02:
##### SUMMARY
In schematic editor, you can add a 1608 resistor or a 2012 resistor.
The package is well received in board editor.
But you can't check which one you have added in the schematic editor. There is nothing in the component properties window (LibrePCB 0.1.2).
You have to get the reference in the schematic editor, go to the board editor, found the device in the list and check the device value.

##### SOLUTION
At least add a field in the component properties window to show the device and package.
And if possible, place a drop-down menu to change the device/package here if needed.
It will be user friendly !

dbrgn, 2019-10-02:
@ubruhin at least the displaying of the chosen device should be easy, right?

ouabache, 2019-10-02:
The schematic doesn't always have that information. If I add a component instead of a device then the schematic doesn't know the device name.

I can have one schematic and two different boards. One thru hole and the other surface mount but the schematic doesn't change.

doom-fr, 2019-10-02:
> The schematic doesn't always have that information. If I add a component instead of a device then the schematic doesn't know the device name.
In this case it can just show 'not specified'

ubruhin, 2019-10-05:
> @ubruhin at least the displaying of the chosen device should be easy, right?

Yeah, probably that should be easy. Even changing the device shouldn't be too hard, so I would say it makes more sense to make it configurable instead of just displaying it.

PR #565 (2019-10-29): Schematic editor: Allow changing pre-selected device of components
Adding a dropdown to the symbol/component properties dialog:

[image]

Fixes #529
closed 2019-10-29 by ubruhin