LibrePCB/LibrePCB #530: Schematic Editor : short pin name & pin details

doom-fr, 2019-10-02:
##### SUMMARY
For some devices like AVR microcontrolers, each pin have multiple functionality and as many names. For devices with a lot of pins, it is a real problem.
The other problem is that components pin names are written in schematic but when we draw the symbol, symbol's names appears so it can produce some problems

Symbol draw : 
[image]

Component result : 
[image]

##### SOLUTION
1 - Show Symbol's pin names in schematics (as explained in doc => Symbol: A symbol is the graphical representation of a Component (or parts of it) in a schematic).
And, as component pin names are useful, put component's pin names in a tooltip or something else.

OR

2 - In component, make 2 different fields 'name' and 'details' and put 'name' field in schematic and 'details' in the tooltip.

ouabache, 2019-10-02:
You can already select between pin,signal or net names to change what is displayed on the schematic. You can also create your own symbol variant and customize your display.

doom-fr, 2019-10-02:
> You can already select between pin,signal or net names to change what is displayed on the schematic.

OK !   Where can I set this ?

ouabache, 2019-10-02:
edit the component
hit slash for the component symbol variant window
the "designator in schematics" defaults to component signal name but has 3 other options

dbrgn, 2019-10-02:
Please see this discussion: https://librepcb.discourse.group/t/alternate-pin-functions/34

doom-fr, 2019-10-02:
Thanks for these information.

ubruhin, 2025-08-24:
I have an idea for this. So far, we have created component signal names like this:

    PB0/RXD/ADC0/INT0

Which is rendered as-is in symbols. We could now introduce `//` as a special separator, like this:

    PB0/RXD//ADC0/INT0

Which would tell LibrePCB to render only `PB0/RXD` by default. The remaining names `ADC0/INT0` could be shown e.g. in the tooltip of the pin. In addition, the pin may provide a context menu like this:

    Displayed Functions:
        ☑ PB0
        ☑ RXD
        ☐ ADC0
        ☐ INT0

Where the user can check/uncheck each name individually, so the schematic editor renders only the checked names (strings joined with `/`, e.g. `PB0/INT0`). In fact I'd implement the file format just as a string like `(override_name "PB0/INT0"`) so actually the UI could even provide the option to manually enter an arbitrary string.

I think it would be a rather simple solution, requiring only two small file format changes:

- Considering `//` as a separator (not really a file format change, just a change of interpretation of existing data)
- Adding some kind of `override_name` property for each pin in a schematic

Open questions:
- Is it too much guideline enforcing? For now, `/` did not have any special meaning for LibrePCB - the whole signal name is just rendered as a complete string, so the user could choose any separator he liked (slash, comma, dash, underscore, ...). With the suggested solution, LibrePCB would thread `/` as a separator, which basically enforces users to exclusively use the slash as separator. Any other separator would prevent this new feature from working.
- Is `//` safe enough that we will never need a signal name containing this string as-is (i.e. not interpreted by LibrePCB)? Do we need to provide an escaping option, e.g. `///` which would be rendered as `//`? Or render `//` as-is and use `///` as the separator?