LibrePCB/LibrePCB #534: Library Editor : set top pins text same orientation as bottom ones

doom-fr, 2019-10-07:
##### SUMMARY & SOLUTION
Set top pins text same orientation as bottom ones, like classic texts
(I know that it is preferable to make functional symbol, but I will do both)

[image]

dbrgn, 2019-10-07:
(Unrelated comment: In case you want to contribute parts to the official libraries, we prefer pin placement only on the sides: https://docs.librepcb.org/#libraryconventions-symbols-pinplacement)

doom-fr, 2019-10-07:
No problem, as explained, I will do both placement type symbol : functional and physical.

dbrgn, 2019-10-07:
Ah, I overlooked that :slightly_smiling_face:

ubruhin, 2019-10-14:
This seems to be a bug in the symbol editor - as soon as the symbol is added to a schematic, rotation is pin texts is correct.

mexchip, 2019-10-19:
I was checking this one, and found that in libs/librepcb/library/sym/symbolpingraphicsitem.cpp the pin is not directly drawn but as different components elsewhere (I didn't found where), while in libs/librepcb/library/sym/symbolpinpreviewgraphicsitem.cpp the _updateCacheAndRepaint_ and _ _paint_ method take into account the orientation, I thought implementing the _paint_ method in the former would fix it but was not able to get it working:

- Trying to edit mTextGraphicsItem rotation and position in symbolpingraphicsitem.cpp is difficult since not able to access several properties (sizes, layers, etc.) directly because not part of the class.
- If trying to implement the _paint_ method in symbolpingraphicsitem.cpp, the drawing duplicates, still.
- If overriding the QGraphicsItem(parent) initializer in symbolpingraphicsitem.cpp with QGraphicsItem() trying to avoid painting twice, the pin is not drawn at all.

I was about to ask for some guidance, then I noticed this other issue: #181, maybe this issue could be deprecated in favor of #181, or still be worked by itself?

ubruhin, 2019-10-19:
I suspect the issue is in `PrimitiveTextGraphicsItem`, which looks a bit ugly anyway :wink::

https://github.com/LibrePCB/LibrePCB/blob/2e747216114042648c9677d7c18301b793b80918/libs/librepcb/common/graphics/primitivetextgraphicsitem.cpp#L134-L145

mexchip, 2019-10-22:
I've made some changes and it seems to work better now. I've finished rebasing so that you can have a look at it.

PR #967 (2022-04-15): Fix rotation & height of some symbol / pin texts
- Fix possibly wrong symbol/pin text auto-rotations in some cases
- Fix inconsistent height of symbol texts between symbol editor and schematic editor

Fixes #534

Replaces #556
closed 2022-04-15 by ubruhin