LibrePCB/LibrePCB #536: Selection of stacked items

doom-fr, 2019-10-08:
##### SUMMARY
In Library Package editor (but maybe elsewhere), when there's different items stacked, you can only select the upper one.

[image]

For example, here, I can't select the Documentation Layer rectangle of the pin as it is completely inside the pad on Top Layer.

I have to move the pad away, select the pin and do what I have to do and move the pad back.

##### SOLUTION
2 different solutions : 
1. when you select a point with stacked items, open a menu to choose the one you need
2. when you select a point with stacked items, firstly select the one that is inside others

Any of this solution will break the problem.
And a mix of both will be perfect !

(layer filtering can also solve this problem but will not be user friendly in Board Editor)

doom-fr, 2020-05-17:
great to be planned in 0.1.5 milestone. :+1:

ubruhin, 2020-08-20:
With #748 this is fixed in the board editor, but other editors still have this issue.

5n8ke, 2020-08-20:
I'm currently working on implementing this for the schematic editor

PR #748 (2020-08-20): Board editor: Refactor and improve context menu
Previously the menu was build and executed for each different item type separately. This resulted in duplicate code and increased effort when adding an already established action to an item. This PR mainly condenses duplicate code and creates an easy framework for new actions.

# Principle
A range of standard actions are already implemented, that can be used for building the context menu (remove, rotate, flip, snap, ...).
If an item needs a custom action, it must be defined locally and a callback must be connected to the QAction::triggered signal [... cut]
PR #757 (2020-08-26): Editors: Improve context menu handling
Improve context menu handling for the symbol, package, and schematic editor.

- Implement improvements as done in #748
- Implement findItemsAtPosition function where necessary
- Allow cycling through items while holding the shift key
- Fixes #536
closed 2020-08-26 by ubruhin