LibrePCB/LibrePCB #539: Board editor : Net clearance

doom-fr, 2019-10-08:
##### SUMMARY
In Board Editor, it seems that net clearance constraints are missing.
:fire::fire::fire: I succeeded in connecting VCC net to GND net ! :fire::fire::fire: 

##### SOLUTION
Implement net clearance, if possible in live.

(Think this is a main pcb software feature that is missing, and so it needs a high priority)

dbrgn, 2019-10-09:
The board editor does not yet have an ERC (Electrical Rules Checker) implemented, that's still one of the big missing features. I'll rename the issue if that's OK for you. (If you meant something different, just leave a comment.)

doom-fr, 2019-10-09:
OK for that renaming.
Just for me, Net clearance is the most important point of ERC.

Is there any workaround till the implementation ?

dbrgn, 2019-10-09:
> Is there any workaround till the implementation?

No, unfortunately not :slightly_frowning_face:

EdizonTN, 2019-10-09:
ERC - electrically rule check - apply in schematic
DRC - design rule check  - apply in PCB editor, includes also clearance/overlay/crossing two separate nets.

I recommend not swap these functions and naming.

doom-fr, 2019-10-09:
https://en.wikipedia.org/wiki/Physical_verification

dbrgn, 2019-10-09:
Good point, sorry :slightly_smiling_face: I always mix up those two.

doom-fr, 2019-10-11:
Thanks to @ubruhin to set this as "High Prio".

I also found this link which can be useful (it is in python but might be easily be translated in C++) : https://stackoverflow.com/a/2824596/1619521

Maybe you have already something like that in LibrePCB as I have found that some clearance checks already exists in library editor : 

[image]

[image]

Maybe this can be spread to DRC ?

ubruhin, 2019-10-11:
I'll try to implement at least some of the most important checks - missing connections and short circuits. Afterwards we can add more checks step by step.

> Maybe you have already something like that in LibrePCB as I have found that some clearance checks already exists in library editor :

Yes, we already have such functions :)

doom-fr, 2019-10-11:
I think that short circuit and clearance is quite the same : 
- short circuit : distance = 0 on different nets
    (for traces : distance between center segments < (width 1+ width 2))
- clearance : distance < clearance on different nets
    (for traces : distance between center segments < (width 1+ width 2 + clearance)

PR #581 (2019-11-25): Board editor: Add initial design rule check
Actually I planned to implement only a very simple DRC for now, but it became much better than expected :grin:

- Add "Design Rule Check" dialog to the board editor, to run the DRC.
- Add "DRC" dock widget to the board editor, to see the messages even after closing the dialog. By clicking on a message, the issue is highlighted and centered on the board view.

[image]

Limitations:
- The "missing connections" check does not really check the copper layers, but only emits a message for each existing airwire. If there is a false-positive airwire, you wil [... cut]
closed 2019-11-25 by ubruhin