LibrePCB/LibrePCB #540: Pckage Editor: Create custom pad shape

EdizonTN, 2019-10-09:
##### SUMMARY
Possibility of creating custom pad shapes

##### SOLUTION
My proposal:
Create a custom pad:
- Place into same copper layer, various object as: polygon, circle, line, hole or Pad. Or combination of these objects of course
- selecting these objects (click LMB + move over objects to select all)
- Click right mouse button - open popup menu - select "Create PAD" item (needs be added)
- now LibrePCB ask to user for selecting center point of this shape.
- After selecting pad center, is custom pad shape created and all objects are "grouped".
- double click on this new shape (custom pad), will open standard pad properties dialog with grayed checkboxes: Shape-Round, Rect and Octagon. Don't add a new shape checkbox "Custom". Can be problem with change from custom to others. Custom shape can be deleted or edit only.

Custom pad (group of objects) is now one object. Can be moved, rotated....

Modify a custom pad:
- RMB on existing custom pad - open popup menu - select "Break up Pad" item (needs be added). "Create Pad" is grayed now.
- and now apply same procedure as create pad.

LibrePCB now works with custom pad as with standard one - create solder mask with positive expansion and paste mask with negative expansion.

Requirements: 
- Minimal one object must be in copper layer (cannot create custom pad only by hole/s)
- If create custom pad with more than one objects, all object must be in same layer 
- Also, selected objects must be overlayed (electrically connected) - minimal is zero distance between objects margins. Otherwise objects will be ignored - even selected too.

Example (mystical):
Used object (Circle, polygon, line with arc filled, hole and std. pad):
[image]

Created custom pad:
[image]

ouabache, 2019-10-14:
You could do this by simply making the footprint editor recursive. In other words you can select other footprints and group them together to create a new footprint.

You would create a set of footprints that are primitives to be used as building blocks that create other footprints. Each one could be rotated and given a unique instance name.

ubruhin, 2023-02-14:
FYI, I implemented custom pad shapes now, but I decided to keep the concept pretty simple for now, see details in #1102. The UI could still be improved some day to make designing custom shapes easier (maybe with grouping objects together as suggested), but I think this should not affect the way how custom shapes are implemented under the hood.

closed 2023-02-16 by ubruhin