LibrePCB/LibrePCB #542: Schematic Editor : show from which lib the component comes (in "Add Component" window)

doom-fr, 2019-10-12:
##### SUMMARY
When you add a component to a schematic using the "Add Component" window, you can't know in which lib it has been found.
And if after putting it on the schematic, there's no field "Library" in "Component Properties" to know this.
In "Library Elements" section of Component Properties", you have a link to the local lib of the project (what's the aim of this ?).
So you have to check every libraries to know from where your component comes.

##### SOLUTION
Add a field 'Library' in the "Add Component" window.
Add the same field in the component properties.

ouabache, 2019-10-12:
What would you do with this information if you did know?

The nice thing about LibrePCB's category system is that you  don't care where the part is stored don't need the library name to find it. The only time you need this information is during development and you want to switch between schematic and library editors.

I think a menu pick to open schematic instance in library editor would be more useful.

doom-fr, 2019-10-12:
1. If I want to update/correct it
2. If I want to fork it
3. If I want to know how one thing have been done in this component
...

ouabache, 2019-10-13:
OK, I hadn't thought about the use case of forking a component. 

You add a component that is "close" to what you really want. You then want to go to that components library and duplicate it into a variant that you can modify to your exact needs.