LibrePCB/LibrePCB #543: Consistent decimal separator, the one of the numeric keypad

doom-fr, 2019-10-12:
##### SUMMARY
First, I am french and have an azerty french keyboard.
My locals define ',' (coma) as decimal separator and a '.' (dot) is written on the numeric keypad (and when I type on it, it usually produces a dot)
The easier when you type decimal numbers is to use the numeric keypad and that the decimal separator of this one is recognized as the decimal separator whatever is your locals.

In Library Editor, the coma is mostly the decimal separator for me, it is not useful (as it is not on the numeric keypad).
An exception is in the "Polygone properties" window which use dot as decimal separator.
This fields are for me the easier to fill.

##### SOLUTION
Use everywhere in LibrePcb the decimal separator of the numeric keypad as decimal separator, or make this one configurable.

dbrgn, 2019-10-12:
Which version of LibrePCB are you using? I think this feature request was implemented in #519.

doom-fr, 2019-10-12:
LibrePCB Version: 0.1.2
Git Revision:     acdd94d9d2310f79215125b999153e9da88a9376
Build Date:       2019-07-13T21:11:00
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.3 LTS
Platform Plugin:  xcb

doom-fr, 2019-10-12:
The problem in Ubuntu is that date format, currency format and decimal separator is bundled.

dbrgn, 2019-10-12:
Please try the [nightly AppImage](https://download.librepcb.org/nightly_builds/master/) to see whether the issue was fixed.

I'll close this issue, but feel free to leave a comment if something's still missing.

closed 2019-10-12 by dbrgn

ubruhin, 2019-10-14:
I think this is the other way around - with #519, the dot will stop working as decimal separator if the locale has defined the comma as decimal separator.

But I wonder if that's normal that with french locale the dot key of the numeric keyboard is not the proper decimal separator? This sounds really strange to me. Do other software accept the dot as decimal separator (e.g. the calculator app or something like that)?

doom-fr, 2019-10-14:
I do not no exactly why but playing with `.profile` and [tweaks](https://askubuntu.com/questions/816610/numeric-keypad-switch-comma-to-period-german-keyboard-layout/1090472#1090472), now it works.

So do not take care of this. I will reopen if the problems comes back.

doom-fr, 2020-06-02:
I have reinstall my pc some month ago and loose the configuration about that.
So now the problem comes back.

What I want to say is that whatever the configuration of Ubuntu is, the decimal separator on the numeric keypad should be recognize as decimal separator.

If you can't fix this (as there might be a lot of different situations), can't you just please consider `.` and`,` as decimal separator in numeric field whatever the configuration of locals is ?

doom-fr, 2020-06-02:
Can you please reopen the issue ?

doom-fr, 2020-06-02:
Seems there's a workaround for me with setting the software to American : 
Control Panel > Extras > Workspace settings > General > Application Language > American English

ubruhin, 2020-06-14:
> If you can't fix this (as there might be a lot of different situations), can't you just please consider . and, as decimal separator in numeric field whatever the configuration of locals is ?

There exist locales which use "," as the thousand separator, thus we shouldn't consider it as a decimal separator too.

I still wonder whether LibrePCB is the only application with this behavior, or if basically *all* applications behave like this(?).