LibrePCB/LibrePCB #544: Schematic Editor : Add Keyboard Shortcuts for main functionalities

doom-fr, 2019-10-13:
##### SUMMARY
In Schematic Editor, "Select", "Draw Wire", "Add NetLabel" and "Add Component" do not have keyboard shortcuts (or it is not showed in the menu).

##### SOLUTION
Add keyboard shortcuts for main functionalities (not only the four I mentioned) and show it in the menu.

For information, in Kicad, if you type `'?'` you'll get all the shortcuts you can use in the current context.
Think the idea is good, but the implementation is bad, as it is never up to date.

EdizonTN, 2019-10-13:
Same as #495

doom-fr, 2019-10-13:
Not the same.

I just need shortcut. Not customised shortcut.

And think that F1 is already used.

EdizonTN, 2019-10-13:
Ok, little bit different :-)
I wrote about user defined shortcut for **all functions** (so, new function also). Of course, all of these have to own default value.
F1: yes, F1 is used for call user's manual on the web. But F1 as main help (without access to web) is most important. I think.

closed 2022-06-16 by ubruhin