LibrePCB/LibrePCB #545: Schematic Editor : Can't add updated component if old version is already in schematic

doom-fr, 2019-10-13:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.3-unstable
Git Revision:     2e747216114042648c9677d7c18301b793b80918
Build Date:       2019-10-10T18:14:18
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.3 LTS
Platform Plugin:  xcb
```

##### SUMMARY
If you add a component in a schematic that already have an old version of the same component, you will get the old one not the updated one.

##### STEPS TO REPRODUCE
Add a component to your schematic.
Update the component (for example change the visibility of a pin name).
Save and wait for end of 'scanning libraries'.
Add the same component (in theory the updated one) to your schematic.

##### EXPECTED RESULTS
The updated version of component is added to your schematic.

##### ACTUAL RESULTS
In fact, the old version of component is added (at least symbol)
(note that in the "Add Component" window, the symbol is the good one).

##### ADDITIONAL INFORMATION
Make "Project" > "Update Library" (think this should be "Update Librairies").
The symbol is now good !

Seems to be the same for devices.

I have experienced many little problems that is difficult to reproduce (so I don't make any issue) and I think that these should come from this problem.

If it is not possible, to have both old and new version in the same schematic, ask the user what to do : 
1. keep the old version
2. update all components of this type in the current schematic

ouabache, 2019-10-13:
If you had duplicated the component before modifying it then this would have worked as you wanted.

dbrgn, 2019-10-13:
I think the reason for this is that the library element is stored in the project cache, and if you re-add the same symbol it will use the cached version.

I like this approach:

> If it is not possible, to have both old and new version in the same schematic, ask the user what to do :
> 
>  - keep the old version
>  - update all components of this type in the current schematic

doom-fr, 2019-10-13:
> I think the reason for this is that the library element is stored in the project cache, and if you re-add the same symbol it will use the cached version.

Yes but when I save the library, it should update the cached version in the project.
And it does, because if you have one component of a certain type in your schematic and add a new one, you will get the old one. But if you then suppress the two components and add one again, it will be the updated one !

NB : I have not changed the field "version" of the component, but it does not change the behavior if it is updated (just tested it).

doom-fr, 2019-10-13:
And also, the symbol preview of the "Add Component" window should be the same of the one that will be added in the schematic.

ubruhin, 2019-10-27:
> Yes but when I save the library, it should update the cached version in the project.

No, that must not happen because it would be very dangerous. For example if your PCB already passed all qualifications, but then you need to add one more device to the PCB (e.g. a resistor), the whole rest of the project must not be touched. It would be horrible if all other resistors on the board would be updated to the latest footprint automatically since this would change the whole design and may create clearance issues etc.

And btw, the project does not have a *cache* - it's the project's *library* - For example when adding a new component to a schematic, LibrePCB copies the component from the workspace into the project. When adding the same component again, it is already available in the project's library, thus that one will be used. When removing components from a schematic, they will be removed from the project library only if they are no longer used somewhere in the project. This concept makes LibrePCB projects 100% independent of the workspace and its libraries.

> Does the actual schematic file format handle the possibility to have two component with the same type but in different versions ?

No, that's not possible right now. I'm also not sure if it makes sense, somehow it feels strange to have e.g. two different versions of 0805 packages on the same board...

> And also, the symbol preview of the "Add Component" window should be the same of the one that will be added in the schematic.

Yes, of course :+1:

doom-fr, 2019-11-14:
Seems there is the same problem with packages in board editor.

It is really a problem. I think that a warning message should at least be pop up.