LibrePCB/LibrePCB #547: Schematic Editor : just first device is good when adding multiple times

doom-fr, 2019-10-14:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.3-unstable
Git Revision:     2e747216114042648c9677d7c18301b793b80918
Build Date:       2019-10-10T18:14:18
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.3 LTS
Platform Plugin:  xcb
```

##### SUMMARY
When adding multiple time the same device, only the first one is good int the "Place Devices" dock the other seems to be the default device of the component.
When clicking "Auto-Add-All", none are good.

##### STEPS TO REPRODUCE
1. Open the "Add component" window
2. select the 1608 resistor device
3. insert it multiple times in the schematic
[image]

4. go to board and checks devices

[image]
***********
[image]
***********
[image]

5. add all devices
=> all are bad ones
[image]

6. remove all devices from the board
=> it goes to step 4
7. now add the first one with "Add" button
=> a 1608 resistor is added
=> now all others have the good device in the "Place Devices" dock
[image]

8. But if you add all of them, none is good
[image]

:angry:

##### EXPECTED RESULTS
All are 1608 resistors

##### ACTUAL RESULTS
Just the first one is 1608.
Others are through hole ones.

##### ADDITIONAL INFORMATION

dbrgn, 2019-10-14:
@ubruhin candidate for 0.1.3?

ubruhin, 2019-10-14:
Indeed, that's not good ;)

PR #561 (2019-10-22): Schematic editor: Fix device when adding component multiple times
When adding a component with a pre-selected device multiple times to a schematic, the device was set only on the first component placed. All successive components didn't have a device selected.

Fixes #547
closed 2019-10-22 by ubruhin