LibrePCB/LibrePCB #548: Allow creating component from symbol

dbrgn, 2019-10-15:
##### SUMMARY

Often you need to create a component that matches a symbol 1:1. This is quite tedious, since all fields need to be copy-pasted-

##### SOLUTION

In the context menu of a symbol, add an entry "Create component". Open the component creation dialog, but pre-fill the following fields, copying the values from the symbol:

- Name
- Description
- Keywords
- Author
- Categories

ouabache, 2019-10-17:
Plus copy all the symbol pins to create the component signals and set the default symbol variant to point back to the symbol.

ubruhin, 2019-10-19:
> In the context menu of a symbol, add an entry "Create component".

That helps only if you want to add the component in the same library as the symbol resides, but often that's not the case.

I'd suggest to just change the order of pages in the component creation wizard. When creating a new component, the wizard could first ask for the symbols to add, and for metadata afterwards. If only one symbol was added, the metadata could automatically be copied from the symbol.

I'm not sure if that's really a good idea, but it might be worth a try. It would also fix another issue: Currently the entered metadata is lost if you first enter the metadata and afterwards realize that the required symbol does not exist yet. The only option is to close the dialog (and thus loose the metadata), create the symbol, and re-open the dialog again afterwards.

dbrgn, 2019-10-19:
> The only option is to close the dialog (and thus loose the metadata), create the symbol, and re-open the dialog again afterwards.

Yep, happened to me very often.

> When creating a new component, the wizard could first ask for the symbols to add, and for metadata afterwards. If only one symbol was added, the metadata could automatically be copied from the symbol.

Would probably be a good idea, yes.

rnestler, 2019-10-19:
> That helps only if you want to add the component in the same library as the symbol resides, but often that's not the case.

Isn't it the 90% case that the component is in the same library?

I'd suggest to do both. Have it select the symbol as the first step in the wizard, or create it directly from the context menu and skip/pre-fill the first step in the wizard.

ouabache, 2019-10-19:
I would suggest we bump that up to 100%. If you create a symbol then you
must create a component for that symbol. This could be automatically done
by making a copy from that symbol.

The reason is that it is currently possible to create a symbol that is only
used in a symbol variant of some other component. It doesn't have to have
ts own component.

So if I have a symbol that is used in variants of 3 other components then I
need to chose a component  when I add it to the schematic. If I change my
mind then I have to delete it and readd the same symbol from a different
component. I don't want to do that. I want to add the symbol and then come
back and change the component that it refers to.

It's the same thing that we do with devices. I can add a component to the
schematic and then later decide if it is thruHole or surface mount without
having to delete and readd.

I want to be able to add a OpAmp and then later decide if it is from a
package of 1,2 or 4 opamps. You can't do that currently but you could if we
have a packing scheme that let you swap between  components That will only
work if every symbol has its own component.