LibrePCB/LibrePCB #549: Board Editor : UI/UX enhancement in drawing wires

doom-fr, 2019-10-16:
##### SUMMARY
UI/UX of drawing wires is a bit rigid and can be easily enhanced.

I think that 98% of wires are sequences of 45° angled segments (between them).
Using these 5 icones are not really user friendly.
[image]
When you draw a wire you want to easily put the wire close to the previous one.

##### SOLUTION
To keep ability to do what we want but with enhancing UI/UX, I propose to do so : 
1. keep only 3 icones : right angles, 45° angles and free lines
2. use 'space' key to switch between : 
     a. in right angle mode : start vertical or start horizontal
     b. in 45° inclined mode : start vertical/horizontal or start 45° inclined
(it seems that right click is doing that withing the five original choices, but it does not really make sense regarding the 98% above, and many time it ends the drawing)
3. when clicking, just register the first half of the current proposition, not both.
It will help to put wire close to the previous one.
4. and so draw the first half of the current proposition should be filled (as now) but the second half only the outline stroke to make comprehensible what will be done (I think it's like that in other softwares).
5. Use 'tab' key to pop a windows with parameters, and first parameter should be the width and the second one the layer (I think it like that in Altium)
6. add a shortcut to draw wire and place via, and document it in tooltip.
7. If you change layer, autoplace a via at the end of the last segment and continue on the new layer

Excepting the last point, I think it is quite easy and really enhanced the user experience.

dbrgn, 2019-10-16:
I think a lot of this would be obsoleted by [integrating the KiCad push & shove router](https://librepcb.discourse.group/t/add-interactive-push-and-shove-router-from-cern-kicad-horizon/13).

The blocker for that is currently someone that would extract the code into a standalone library and maintain that. Anyone interested? :slightly_smiling_face:

doom-fr, 2019-10-16:
Ok, push and shove router is cool.

But it is not integrated since quite 2 years now and no one is in charge of that.
So to help current users of LibrePCB to wait for this integration, I think it will be cool to integrate my enhancement propositions now (at least the 6 first points as they are quite easy I think).

I also use this message to add a point 8 and precise/correct the point 5 and 6 : 
8. add a clearance area (same copper layer color with some transparency) to see what we are routing like in kicad
5. it is not necessary to pop a window, but just to go with 'tab' to the already existing fields beginning by the track width (which is the most used) and using 'tab' to navigate properly in these fields.
6. I think it will be cool to add a shortcut for top wire and another for bottom (and maybe for internal layers also).

And for the future, when push and shove router will be integrated one day, does it mean that you will take the User Interface of Kicad or just the push & shove algorithm ?
And will it change the board file format ?
If yes will you provide an old-to-new transcription tool ?

Thanks.

ubruhin, 2019-10-22:
> So to help current users of LibrePCB to wait for this integration, I think it will be cool to integrate my enhancement propositions now (at least the 6 first points as they are quite easy I think).

The main problem is that the board editor logic to draw wires is already quite complex (and not very well implemented) and it will take quite some time to change the behavior and to test it intensively. Thus it might not be worth the effort to change it, compared to the effort needed to integrate the push&shove router. For example, improving the current router might cost 5 days and increases usability by ~20%, while the push&shove router might cost 20 days but increases usability by >500% ;)

> And for the future, when push and shove router will be integrated one day, does it mean that you will take the User Interface of Kicad or just the push & shove algorithm ?

Of course only the push&shove algorithm, the user interface will still be from LibrePCB.

> And will it change the board file format ?

Probably not, at least not at the beginning. Maybe some day we want to store additional settings for the push&shove router.

>  If yes will you provide an old-to-new transcription tool ?

If we make changes to the file format, of course we will provide an upgrade mechanism (see versioning concept [here](https://developers.librepcb.org/d1/d69/doc_versioning.html)).

doom-fr, 2023-08-27:
This issue is still needed as the current trace drawing is not easy to use

doom-fr, 2023-08-28:
related to https://github.com/LibrePCB/LibrePCB/issues/600