LibrePCB/LibrePCB #550: Bug : can't start bottom copper wire at end of top copper wire (even with via)

doom-fr, 2019-10-16:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.3-unstable
Git Revision:     2e747216114042648c9677d7c18301b793b80918
Build Date:       2019-10-10T18:14:18
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.3 LTS
Platform Plugin:  xcb
```

##### SUMMARY
It is not possible to draw bottom wire at the end of a top wire.
It automatically switch to top wire, even if there is a via at this place.

##### STEPS TO REPRODUCE
1. draw a  top wire
2. place a via at the end
3. try to draw a bottom wire

##### EXPECTED RESULTS
draw a bottom wire

##### ACTUAL RESULTS
draw a top wire

##### ADDITIONAL INFORMATION

dbrgn, 2019-10-16:
Workaround for this: Start connecting from the oposite side *to* the via, instead of starting at the via.

doom-fr, 2019-10-16:
Think that it does not work if you want to do top / bottom / top wire, which is usual in CMS board.

doom-fr, 2019-10-16:
The workaround that works is to put the via alone, to connect it with bottom wire and after to make the top wire starting from the top cms device.
But, you understand that it is not really user friendly.

dbrgn, 2019-10-16:
Yep, it's definitely a bug :slightly_smiling_face:

doom-fr, 2020-06-07:
Seems that this bug occurs more when there's a very little piece of wire just above the via.
It seem it focus on the node of the wire (which is on top), instead of the via.
The workaround is to move the cursor in the opposite side of the via to make the center of via closer than the node.

5n8ke, 2020-08-31:
@doom-fr Was this fixed by #733

ubruhin, 2020-09-13:
Should be fixed with #733, thus closing this issue. Feel free to reopen if the next LibrePCB release does not fix the problem.

closed 2020-09-13 by ubruhin