LibrePCB/LibrePCB #552: Schematic Editor : Bug : crash (Segmentation Fault) when adding component and typing 'esc' 2 times

doom-fr, 2019-10-18:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.3-unstable
Git Revision:     2e747216114042648c9677d7c18301b793b80918
Build Date:       2019-10-10T18:14:18
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.3 LTS
Platform Plugin:  xcb
```

##### SUMMARY
LibrePCB crashes when adding a component with the 'add component' window
and typing 2 times 'esc'

##### STEPS TO REPRODUCE

1. open the 'add component' window 
2. place a component
3. type 'esc' to return to the 'add component' window
4. type 'esc' to return to main schematic window

##### EXPECTED RESULTS
Just return to schematic main window

##### ACTUAL RESULTS
LibrePCB crashes with segmentation fault

```
[DEBUG-MSG] Workspace library scan succeeded: 1845 elements in 13991 ms (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Autosave project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully autosaved (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] Workspace library scan started. (:0)
[DEBUG-MSG] Workspace libraries indexed: 8 libraries in 15 ms (:0)
[DEBUG-MSG] Workspace library scan succeeded: 1845 elements in 13904 ms (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Autosave project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully autosaved (:0)
[DEBUG-MSG] Autosave project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully autosaved (:0)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Autosave project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully autosaved (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] Autosave project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully autosaved (:0)
[DEBUG-MSG] Autosave project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully autosaved (:0)
Segmentation fault (core dumped)

```

##### ADDITIONAL INFORMATION
It is not systematic.
Some days before it happens a lot when I was dealing with adding components
to my local lib and inserting them in my schematic.
So I decided to launch LibrePCB from console to have logs.
And nothing for many days ...
Until now !

doom-fr, 2019-10-18:
Another crash

```
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] Request successfully finished: "https://api.librepcb.org/api/v1/libraries/v0.1" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/LibrePCB_Connectors.lplib/bc15eae0852a3518122c7020da162953adee09f5/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/LibrePCB_Base.lplib/999a77172ccd62e93c6002055156131e059a9317/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/STMicroelectronics.lplib/7645fc98afd54826e48420cd131920bd2f0b0a93/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/Isabellenhuette.lplib/241938a2c1e44604165b8e7ae9bc6d5bac7c810f/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/Molex.lplib/4c06c52c699c0d7d817f0c630a539606e02098c7/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/LibrePCB_ICs.lplib/9181471c673a1744d34ccb885f34ca62aa4f9315/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/Microchip.lplib/441191a0c2ab009011c4be2ef10d9192d73f6d32/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/Arduino.lplib/011dbdf716e3647aba44d4788c4467ad84298b4e/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/Sensirion.lplib/468971cbe3c1f20820d4dffb16900da7492ce654/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/Luminus.lplib/cef5532c8ca44e8a3015955c7b9aa6b558762cb6/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/Maxim.lplib/f541226685c901936eea92a33282c450e0865695/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/Diodes_Incorporated.lplib/3b9c105214542c1863a18e3a9d2d42e378eb0da6/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/FTDI.lplib/812ee82afd5385da9eb840228f899311863e4b3f/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/Texas_Instruments.lplib/6c81f0f4924a2bc5f84a61f4f795dec76c6dfd2b/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/OSRAM.lplib/b53cf17abdad8ee0c970adb2229b17d1f0c96195/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/KEMET.lplib/3afbd01e8cd54d7e699c19c35754c45e3866c2d2/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/Panasonic.lplib/eb3c2994de823f4feee863690518090ed7ad2f1d/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/Toshiba.lplib/aebc203c74b3e6dbe29d511993e6a49b540bd188/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/Adafruit.lplib/b33347b27bc83adf1721ad36b1d1413f22aed2c8/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/CNC_Tech.lplib/3d08553f18dc6ff0fa9a72f5d0c62a8b6639ad85/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/Keystone.lplib/fc80a04916a284b07d976b9c2daa276728ac94d5/library.png" (:0)
[DEBUG-MSG] Request successfully finished: "https://raw.githubusercontent.com/LibrePCB-Libraries/Cinch_Connectivity_Solutions.lplib/50a2d3dbee489f2265692e1fc0fe3fc35a58144b/library.png" (:0)
[DEBUG-MSG] Workspace library scan started. (:0)
[DEBUG-MSG] Workspace libraries indexed: 8 libraries in 21 ms (:0)
[DEBUG-MSG] Workspace library scan succeeded: 1845 elements in 15163 ms (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Autosave project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully autosaved (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] Save project... (:0)
[... cut ...]
[DEBUG-MSG] Workspace library scan succeeded: 1845 elements in 14750 ms (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] Autosave project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully autosaved (:0)
Segmentation fault (core dumped)
```

dbrgn, 2019-10-18:
I can't reproduce this here. Can you try to create a core dump? https://askubuntu.com/questions/966407/where-do-i-find-the-core-dump-in-ubuntu-16-04lts

doom-fr, 2019-10-18:
done the command `sudo service apport start`, but seems it was already active.

```
doom@doom-PC:~$ sudo service apport status
[sudo] password for doom: 
● apport.service - LSB: automatic crash report generation
   Loaded: loaded (/etc/init.d/apport; generated)
   Active: active (exited) since Fri 2019-10-11 23:49:47 CEST; 6 days ago
     Docs: man:systemd-sysv-generator(8)
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/apport.service

oct. 11 23:49:47 doom-PC systemd[1]: Starting LSB: automatic crash report genera
oct. 11 23:49:47 doom-PC apport[1248]:  * Starting automatic crash report genera
oct. 11 23:49:47 doom-PC apport[1248]:    ...done.
oct. 11 23:49:47 doom-PC systemd[1]: Started LSB: automatic crash report generat
doom@doom-PC:~$ 
doom@doom-PC:~$ 
doom@doom-PC:~$ 
doom@doom-PC:~$ sudo service apport start
doom@doom-PC:~$ 
doom@doom-PC:~$ 
doom@doom-PC:~$ 
doom@doom-PC:~$ sudo service apport status
● apport.service - LSB: automatic crash report generation
   Loaded: loaded (/etc/init.d/apport; generated)
   Active: active (exited) since Fri 2019-10-11 23:49:47 CEST; 6 days ago
     Docs: man:systemd-sysv-generator(8)
    Tasks: 0 (limit: 4915)
   CGroup: /system.slice/apport.service

oct. 11 23:49:47 doom-PC systemd[1]: Starting LSB: automatic crash report genera
oct. 11 23:49:47 doom-PC apport[1248]:  * Starting automatic crash report genera
oct. 11 23:49:47 doom-PC apport[1248]:    ...done.
oct. 11 23:49:47 doom-PC systemd[1]: Started LSB: automatic crash report generat
```

And do not found LibrePCB core in `/var/crash`, only 3 files related to another crash `_usr_bin_gnome-tweaks.1000.crash`

The content of `/var/log/apport.log`

```
ERROR: apport (pid 24140) Fri Oct 18 09:02:08 2019: called for pid 11917, signal 11, core limit 0, dump mode 1
ERROR: apport (pid 24140) Fri Oct 18 09:02:08 2019: Unhandled exception:
Traceback (most recent call last):
  File "/usr/share/apport/apport", line 483, in <module>
    exe_mtime = os.stat('/proc/%s/exe' % pid).st_mtime
PermissionError: [Errno 13] Permission denied: '/proc/11917/exe'
ERROR: apport (pid 24140) Fri Oct 18 09:02:08 2019: pid: 24140, uid: 1000, gid: 1000, euid: 0, egid: 0
ERROR: apport (pid 24140) Fri Oct 18 09:02:08 2019: environment: environ({})
ERROR: apport (pid 26601) Fri Oct 18 10:43:28 2019: called for pid 24268, signal 11, core limit 0, dump mode 1
ERROR: apport (pid 26601) Fri Oct 18 10:43:28 2019: Unhandled exception:
Traceback (most recent call last):
  File "/usr/share/apport/apport", line 483, in <module>
    exe_mtime = os.stat('/proc/%s/exe' % pid).st_mtime
PermissionError: [Errno 13] Permission denied: '/proc/24268/exe'
ERROR: apport (pid 26601) Fri Oct 18 10:43:28 2019: pid: 26601, uid: 1000, gid: 1000, euid: 0, egid: 0
ERROR: apport (pid 26601) Fri Oct 18 10:43:28 2019: environment: environ({})

```

doom-fr, 2019-10-21:
Ok, I have got one core.
How do you want me to send it to you ? (~ 250MB)

doom-fr, 2019-10-21:
```
[DEBUG-MSG] 1 boards successfully loaded! (:0)
[DEBUG-MSG] project successfully loaded! (:0)
[DEBUG-MSG] Workspace library scan succeeded: 1862 elements in 21503 ms (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[EXCEPTION] User Canceled (../../../../libs/librepcb/projecteditor/schematiceditor/fsm/ses_addcomponent.cpp:391)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
[DEBUG-MSG] Autosave project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully autosaved (:0)
[DEBUG-MSG] Save project... (:0)
[DEBUG-MSG] Save project files to transactional file system... (:0)
[DEBUG-MSG] Project successfully saved (:0)
Segmentation fault (core dumped)

```

dbrgn, 2019-10-21:
You could upload it to https://send.firefox.com/ and send the link to @ubruhin (see git log for his e-mail). Don't post it here, since the link expires after a few downloads (and the dump might contain confidential data).

doom-fr, 2019-10-21:
Done !

ubruhin, 2019-10-22:
Thanks for sending the core dump! I tried to load it, but I suspect it is useless if I don't have the debug symbols of the executable :sob:

I also tried to reproduce this issue but without success. @doom-fr did you find a way how to reproduce it, or does it still happen very rarely? When does it crash exactly, immediately when pressing 'esc' the second time?

doom-fr, 2019-10-22:
> Thanks for sending the core dump! I tried to load it, but I suspect it is useless if I don't have the debug symbols of the executable 

Should I do something to have this ? How can I help you ?

> I also tried to reproduce this issue but without success. @doom-fr did you find a way how to reproduce it, or does it still happen very rarely?

It is not systematic. When I was creating my components, it happens a lot.
Now less. So I think there is a relation with that or with the management of the library update.
But I am not sure as it is not systematic.

> When does it crash exactly, immediately when pressing 'esc' the second time?

Yes exactly after the second 'esc' pressing.
What is done at this moment ??

ubruhin, 2019-10-30:
> Should I do something to have this ? How can I help you ?
Should I use the new nightly AppImage ? Do you have the debug symbols of this executable ?

No, unfortunately we don't have debug symbols for the releases and nightly builds. I tried to automatically build them on CI, but I'm not really happy with it because the debug symbols are huge (many hundreds of megabytes). This even slows down CI for several minutes just for uploading them to our server :(

So probably the only way is to compile LibrePCB by yourself (including debug symbols) and then send us both together, the executable and the core dump. I agree that's cumbersome, but at this time I don't have a better idea :(

> Yes exactly after the second 'esc' pressing.
What is done at this moment ??

Actually nothing special, the dialog object gets destroyed and the "Place device" tool is exited...

doom-fr, 2020-06-03:
Playing with last nightly build of branch 0.1 for several days now and not seen this bug again.
Seems it has been fixed !
So I close the issue.

doom-fr, 2020-06-28:
Just experienced this crash !
So if fact still there.

doom-fr, 2020-06-28:
I think this appears when you first add in your schematic a just created component.

doom-fr, 2021-03-11:
I think this is related to https://github.com/LibrePCB/LibrePCB/issues/840
And so fixed with https://github.com/LibrePCB/LibrePCB/pull/855

So I close this for now.

closed 2021-03-11 by doom-fr