LibrePCB/LibrePCB #554: Grid Offset

dbrgn, 2019-10-19:
### SUMMARY

Sometimes packages have pads that are evenly spaced, but that are not on the same grid as other pads. This means that when we align one of the headers to the grid, the others are not aligned, and vice versa.

##### SOLUTION

Allow specifying an x/y offset for the grid. If specified, the entire grid is shifted by that amount.

doom-fr, 2020-05-30:
Maybe easier (regarding user experience) to snap the pad if the cursor is on it, even if not on the grid.

EdizonTN, 2020-12-27:
Specifying of the x/y offset of the grid is good but its needs to know x/y coordination of the pad (or others elements). 
Better solution is combination of the Snap function (snap to pad center, snap to element center/middle point,...) with set start point of the grid.

Set start point of the grid - after calling this function is start of x/y coords of the grid set to actual mouse pointer (after LMB click).

Snap function - is good if is enabled continously during design PCB.
By the shortcut keys, can be enabled/disabled active snap elements:
- grid
- center of hole (pad, via, hole)
- center of endpoint of the track (on any layer)
- middle point of the track element
- start point of the arc
- center of the arc
- angle step of the arc track

Snap function will be a big help for the create a board outline, absolute component placement to the board, routing, create a footprint etc.

Tubbles, 2020-12-29:
I started looking into this

> Maybe easier (regarding user experience) to snap the pad if the cursor is on it, even if not on the grid.

[image]

Already seems to work if I understood your question correctly? :)