LibrePCB/LibrePCB #557: .clang-format file does not work with llvm version > 6

mexchip, 2019-10-20:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.3-unstable
Git Revision:     2e747216114042648c9677d7c18301b793b80918
Build Date:       2019-10-17T01:29:38
Qt Version:       5.13.1 (built against 5.13.1)
CPU Architecture: x86_64
Operating System: Arch Linux
Platform Plugin:  xcb
```

##### SUMMARY
Included clang-format file throws an error (unknown key 'Delimiter') when trying to be run with clan-format version > 6

##### STEPS TO REPRODUCE

1. Clone repository
2. Try to run clang-format on any file
```
$ clang-format -style=file -i apps/librepcb/main.cpp
```
3. Following error is thrown:
```
YAML:100:22: error: unknown key 'Delimiter'
  - Delimiter:       pb
                     ^~
```

##### EXPECTED RESULTS
No error thrown and file formatted according to clang-format file specification.

##### ACTUAL RESULTS
Unkown key 'Delimiter' error thrown.

##### ADDITIONAL INFORMATION

- Similar issue mentioning change of behavior from llvm 6 to 7: https://github.com/QMCPACK/qmcpack/issues/794#issuecomment-390332175
- Current clang-format docs use _Delimiters_ key instead of _Delimiter_: https://clang.llvm.org/docs/ClangFormatStyleOptions.html

mexchip, 2019-10-20:
Current Ubuntu LTS (18.04) uses clang 6 by default, so I guess this is not an important issue right now.

mexchip, 2019-10-20:
I was eager to make it work and edited the file accordingly (so the pull request), but maybe the the RawStringFormats/TextProto section isn't even needed?

dbrgn, 2019-10-20:
You should use the `--docker` flag, which runs clang-format in a Docker container with the correct version.

Unfortunately clang-format is not backwards-compatible, which means that we need to pick a version and stick with it. If the version is too new, then it won't work on some older development systems.

Maybe we could show a hint about the `--docker` flag if clang-format fails...

dbrgn, 2019-10-20:
#559 should probably fix this issue. @mexchip does this work for you?

mexchip, 2019-10-20:
I originally tried to execute _clang-format_ only for one file, but #559 works for me, thanks!

The only thing I'm still confused is what _TextProto_ is useful for and if it is really needed in the .clang-format file.

dbrgn, 2019-10-20:
> The only thing I'm still confused is what TextProto is useful for and if it is really needed in the .clang-format file.

I'm not sure, @ubruhin?

In any case, compatibility between different versions of clang-format is unfortunately not guaranteed, so it wouldn't even help if we'd remove it.

mexchip, 2019-10-21:
If compatibility can't be guaranteed even if removing that section, I think the best option is #559, I'm closing this. Thanks a lot for your help!

mexchip, 2019-10-21:
I've tried removing the section just as this other project did: https://github.com/QMCPACK/qmcpack/commit/a1b95fe3b51e1028ecd76653b417d04cdf8cd318#diff-88d5907f11b8b6ca0b213de568bf4582
and tested successfully with two clang-format versions: 6 (running dev/format_code.sh --docker) and 9 (installed in my machine). So if we could confirm if RawStringFormats/TextProto section is really needed, it could be anotherr option.

ubruhin, 2019-10-21:
> > The only thing I'm still confused is what TextProto is useful for and if it is really needed in the .clang-format file.
> 
> I'm not sure, @ubruhin?

I suspect this section is not needed. But it seems to help enforcing a specific version of `clang-format` :grin:

> and tested successfully with two clang-format versions: 6 (running dev/format_code.sh --docker) and 9 (installed in my machine).

The compatibility issue is not only the `.clang-format` file, but also how the code will be formatted. In my experience, different versions of `clang-format` produce different output. Thus I'd like to *always* use `clang-format` version 6, otherwise the format of files changes every time a developer uses a different `clang-format` version.

You could run `format_code.sh` with the `--all` flag to check if the different versions produce different output (without `--all`, only modified files are formatted, thus not suitable to test formatting).

mexchip, 2019-10-21:
> The compatibility issue is not only the `.clang-format` file, but also how the code will be formatted. 

Ok, I get it now, makes lots of sense.

Running format_code --docker --all on current _master_ branch:
```
$ dev/format_code.sh --docker --all
Formatting files with clang-format...
[Re-running format_code.sh inside Docker container]
Formatting files with clang-format...
Finished: 3 files modified.
Sort *.pro file entries...
Finished: 0 Files modified.
[Docker done.]
```

Without docker (clang-format version 9):
```
dev/format_code.sh  --all
Formatting files with clang-format...
Finished: 13 files modified.
Sort *.pro file entries...
Finished: 0 Files modified.
```

So many differences between clang-format versions. Closing this one again :P.

PR #559 (2019-10-21): If clang-format fails, show hint about the --docker flag
Fixes #557, probably replaces #558.
closed 2019-10-21 by mexchip