LibrePCB/LibrePCB #560: Schematic Editor : Bug : {{DEVICE}} placeholder is blank

doom-fr, 2019-10-20:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.3-unstable
Git Revision:     2e747216114042648c9677d7c18301b793b80918
Build Date:       2019-10-10T18:14:18
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.3 LTS
Platform Plugin:  xcb
```

##### SUMMARY
I have created an Unidirectional Schottky Device and add it to my schematic.

The problem is that the value field is blank on the schematic page.
Value of this component is defined as `{{PARTNUMBER or DEVICE}}`

[image]

It seems that `{{PARTNUMBER}}` and `{{DEVICE}}` fields are blank so it display nothing.
If I put `{{PARTNUMBER or DEVICE or COMPONENT}}`, I have the component name not the `{{DEVICE}}` or `{{PARTNUMBER}}` value.

It seems that in some cases after doing something, the good value comes back.
(a restart of LibrePCB is not enough)

##### STEPS TO REPRODUCE

1. place a new device
2. look at the value field

##### EXPECTED RESULTS
see `{{PARTNUMBER}}` or `{{DEVICE}}` field in value place.

##### ACTUAL RESULTS
see `{{PARTNUMBER}}` instead.

##### ADDITIONAL INFORMATION

dbrgn, 2019-10-20:
This is a known bug, the value will appear as soon as you have placed a device on the board.

You're right that it should already show the value when you've preselected a device in the schematic editor.

doom-fr, 2021-02-08:
It would be nice to fix this bug as it also impacts the BOM generation.

ubruhin, 2024-10-17:
I think this should be better now in LibrePCB 2.x, thus closing. Feel free to reopen if it is still relevant.

closed 2024-10-17 by ubruhin