LibrePCB/LibrePCB #572: Library Editor : changing pin name when placing a pin

doom-fr, 2019-11-09:
##### SUMMARY
When you place a pin in the Library Editor, typing something on keyboard should change the pin name or at least typing 'tab' should highlight the field and allow it.
Currently, it does nothing and typing 'tab' goes to 'name', 'description', ... fields.

##### SOLUTION
Just use 'tab' to rotate through 'name' and 'length' fields

##### version
LibrePCB Version: 0.1.3-unstable
Git Revision:     34e296d0bd32946b5abdb48d178b3507bd996c7f
Build Date:       2019-11-07T22:06:40
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.3 LTS
Platform Plugin:  xcb

doom-fr, 2020-06-02:
When placing a pin, 'tab' should rotate through fields on the command bar.

This behavior should be generalized to all LibePCB.

Tubbles, 2020-12-27:
I would say that this is an important quality of life improvement

ubruhin, 2022-09-15:
This has been implemented in #986 (using the 'Tab' key) :rocket:

closed 2022-09-15 by ubruhin