LibrePCB/LibrePCB #575: Board Editor : 'Auto-Add All' places bad devices on board

doom-fr, 2019-11-10:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.3-unstable
Git Revision:     34e296d0bd32946b5abdb48d178b3507bd996c7f
Build Date:       2019-11-07T22:06:40
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.3 LTS
Platform Plugin:  xcb
```

##### SUMMARY
When adding devices to board with 'Auto-Add All' button, all are bad.
When adding by hand with 'Add' button, it is ok.
But I have more than one hundred devices, so would like to use the 'Add All' function.

##### STEPS TO REPRODUCE
1 . click on the 'Auto-Add All' button

##### EXPECTED RESULTS
All device have good footprint

##### ACTUAL RESULTS
All devices have bad footprint

[image]

[image]

##### ADDITIONAL INFORMATION

dbrgn, 2019-11-10:
What do you mean with "bad footprint"?

doom-fr, 2019-11-11:
For example, Ra2 should be 1608 SMD package (see first picture) and is instead through all package (see second picture).
So understand 'package' instead of 'footprint'.
But so, footprints are also bad ....

doom-fr, 2021-02-08:
Up !
Is it possible to add this bug to the next release ?
It is really annoying, and it seems to be 'easy'. Am I wrong ?

closed 2021-06-14 by ubruhin