LibrePCB/LibrePCB #576: BOM generator doesn't show good device and package

doom-fr, 2019-11-11:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.3-unstable
Git Revision:     c292ef00b9c5be06fb0b67b056d9a924cfbe92b0
Build Date:       2019-11-11T19:57:35
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.3 LTS
Platform Plugin:  xcb
```

##### SUMMARY
BOM generator doesn't show good device and package

##### STEPS TO REPRODUCE
1. generate BOM
2. compare with device in schematic

##### EXPECTED RESULTS
same device and package

##### ACTUAL RESULTS
bad device and package

[image]

[image]

##### ADDITIONAL INFORMATION
Should be related to https://github.com/LibrePCB/LibrePCB/issues/575

doom-fr, 2019-11-11:
Should be related to https://github.com/LibrePCB/LibrePCB/issues/575

doom-fr, 2021-03-11:
I don't understand why device definition is not linked between Schematic and Board Editor.
At least, a warning should be raised when devices are different > https://github.com/LibrePCB/LibrePCB/issues/584

ubruhin, 2024-02-19:
I think this is no longer relevant since LibrePCB 1.0 because device selection now works in a very different way, so let's close it. Feel free to reopen and clarify if it's still relevant.

closed 2024-02-19 by ubruhin