LibrePCB/LibrePCB #578: Board Editor : board outline polygone 'filled' tag is not removed

doom-fr, 2019-11-14:
##### VERSION / OS / ENVIRONMENT
```
LibrePCB Version: 0.1.3-unstable
Git Revision:     c292ef00b9c5be06fb0b67b056d9a924cfbe92b0
Build Date:       2019-11-11T19:57:35
Qt Version:       5.12.3 (built against 5.12.3)
CPU Architecture: x86_64
Operating System: Ubuntu 18.04.3 LTS
Platform Plugin:  xcb
```

##### SUMMARY
If you draw a board outline, set the 'filled' tag, remove it, it is not really removed.

##### STEPS TO REPRODUCE

1. draw a closed board outline
2. click down in the middle and try to move, it does not work (expected)
3. add the 'filled' tag in the properties window
4. try step 2 again, it moves (expected)
5. remove the 'filled' tag
6. try step 2 again, it moves also, which is not expected

##### EXPECTED RESULTS
see above

##### ACTUAL RESULTS
see above

##### ADDITIONAL INFORMATION
no

ubruhin, 2019-11-16:
Thanks for the good explanation! I could reproduce it, found the issue and will fix it.

doom-fr, 2019-11-16:
Great !

PR #582 (2019-11-17): Fix possibly wrong grab area of various graphics items
The shape was sometimes not calculated properly in `Toolbox::shapeFromPath()`, which lead to wrong grab areas of polygons and probably other graphics items, especially after modifying their properties.

Fixes #578
closed 2019-11-17 by ubruhin