LibrePCB/LibrePCB #580: Board Editor : no showing net name on little pads

doom-fr, 2019-11-15:
##### SUMMARY

On too little pads, the net name is not shown.

[image]

##### SOLUTION

reducing the size of the font on little pads

doom-fr, 2019-11-15:
And sometimes net names are cut

[image]

EdizonTN, 2019-11-16:
Change font size depends on Pad size on screen (zoom in/out)?

doom-fr, 2019-11-16:
Sometimes there's enough place to write the name without changing the font size.
So first, should modify the name outbound.
And if it is not sufficient, effectively, shrink it to fit inside.

ubruhin, 2023-04-15:
I think this has been fixed with #1136 (font size gets reduced to always fit into the pad) so let's close this :muscle:

closed 2023-04-15 by ubruhin