LibrePCB/LibrePCB #584: Schematic and Board Editor : set warning if device is different between schematic and board

doom-fr, 2019-11-18:
##### SUMMARY

There seems to be two ways of working with schematic/board editors : 
1. setting devices in the schematic (and so packages) before starting the board
2. setting just components in schematic and set devices in Board Editor.

I you work with the first way (which is the one I use), you can get differences between the one set on the board and the one in the schematic.

##### SOLUTION
Add a warning if the board and schematic devices are different.

doom-fr, 2019-11-18:
I think that for this issue and that one https://github.com/LibrePCB/LibrePCB/issues/585, a new panel with a listing of placed devices should be added.

ouabache, 2019-11-18:
Another way to handle this is to not store the device info in the schematic, only store the component.
If you select a device for the schematic then store the component in the schematic and the device in the board file. That way if you set the device in the board then you are simply changing it rather than having two copies.

doom-fr, 2019-11-18:
Yes, but for example capacitors have different size regarding to capacity, voltage, technology, ...
It depends on things that you can see on the schematic, not on the board.
So setting this on board editor is not easy.

doom-fr, 2020-06-05:
This is really a problem. LibrePCB retain bad component and not the updated one.
It also produces errors in the BOM.

I think this is a bug.

> Of course all of the existing boards are always in sync with the schematics

[image]

Can't we at least add a warnings in the schematic ERC or elsewhere ?

doom-fr, 2020-06-05:
Is there a workaround to know which one is good and which one is bad, without manually verify everything (which is a lot of work and a source of errors).

doom-fr, 2020-06-05:
Seems that in fact BOM is ok.
But BOM is not in sync with fabrication files which retain bad components.

PR #1125 (2023-03-09): DRC: Warn about devices which differ from schematic
Warn about devices which are different to the device configured in the schematics. For example when adding a capacitor with 0603 package to the schematics but then using 0805 in the board, the warning appears.

Closes #584
closed 2023-03-09 by ubruhin